gnea/grbl #889: PWM out of CNC controller not functioning

timtiernan, 2020-07-08:
Hi, I've tried everything I can think of but I'm really stuck! You're my last hope.

I have recently purchased a new CNC board (https://www.amazon.co.uk/dp/B07D211NKD/ref=cm_sw_r_cp_apa_i_FC5aFbWY4T8NA) and laser (https://www.amazon.co.uk/dp/B07VD47BMX/ref=cm_sw_r_cp_apa_i_cD5aFbKE41RB9) but they just will not talk to eachother.

I've got it wired as shown on the page for the board but for some reason when I send the following commands the laser doesn't switch on (PWM is enabled in settings).

[image]

I've tried it with and without the 12v external power to the driver board (adapter purchased separately (https://www.amazon.co.uk/dp/B017OMEW4Q/ref=cm_sw_r_cp_apa_i_RD5aFbARDEXH6)) but I'm not getting anything except the fans on the board and laser running when switched on (so I know it's getting power). I've also tried it with the jumper set to both laser and spindle with no effect. 

I don't believe that the board is faulty as it has s responding to LaserGRBL, just not doing anything on the other end.

I have had CNC boards and lasers work in the past but for some reason this one won't play ball.

Thanks in advance for any help offered.

[image]

[image]

[image]

inee-gas, 2020-08-17:
Hi 
Looking at your pics ,it looks like you have the red and the white wires in the wrong order, as an image of your board shows it as follows
12v   Signal     Gnd
Red   White   Black

just double check it cheers

quotelawrence, 2020-12-27:
I am not sure if this is a PWM issue though I just had to replace a laser module the wiring is not the same as the last so I am attempting to find hot and ground and ttl/a the image I have seen was red hot, black ground and blue was ttl/a, but when I touch hot lead to red and black to ground I get the laser on no adjustment just burning, I touch ground to the TTL/A it clicks as if a solenoid is firing, but when I hook up everything the table moves fantastically, but the laser never turn on so is this due to firmware, or wiring or both this is driving me nuts

inee-gas, 2020-12-28:
Hi
Not sure if thats your board but, if you look at the middle step board, on the main pcb is a red jumper,in the image its set to spindle ,looks like the jumper needs moved to laser