gnea/grbl #890: Hi. How can I generate gcode from dxf? Is there any free software?

Helio12, 2020-07-14:


uChip, 2020-07-14:
The free version of Carbide Create can import dxf's.  So can Fusion 360.  In either case you will need to do some work to turn the dxf into gcode.

MeJasonT, 2020-07-14:
Its a strange concept and not entirely what you would expect.  Free does not mean free.
I had never heard of Carbide create so just had to pull it up and have a look,  On an entry level for free it looks pretty cool.
so my two suggestions would be.
Cambam, I found Cambam to be a very easy and intuitive program to get started with, there is a free version floating about but is limited to the length of code.  There is a modest charge which isnt unreasonable, Its no where near the price of Mach3 (which is a machine GUI - just making the price comparison, its 3x more cost than Cambam).  Cambam is great and will produce some good cuts but its tool paths are limited - not a lot of choice ie no adaptive paths.
Fusion360 is all singing and dancing and although hobby people are not charged a fee, dont make the mistake it will always be free- Autodesk are quite explicit and quite within their rights to remove the free of charge product.  I imagine in the future it will go pay per view (as it were).
So conFusion 360,  Learning or indeed switching to Fusion 360 is not an easy move, steep learning curve but there is loads of material available to teach the basics.  

The guy i like is [https://www.youtube.com/channel/UCo29kn3d9ziFUZGZ50VKvWA](url)

Lars works for autodesk now and explains things in really simple language, can some times come across as a little drawn out but actually its the small details you think you know that will trip you up.  Its like forgetting to put line terminators when you are programming ":".

The good thing about the free loaned licence fron autodesk is that you can do some real wizzy tool paths and really get the best out of your milling.  
The comparison between cambam and fusion is that cambam will either profile or pocket, there is a trochoidal mill path but it really just pocketing in circles.  Fusion mixes the paths with the adaptive tool path and will simutaniously give you all three milling operations at the same time, thus speading up your milling job and inproving your finish.

So the entry level isnt free but a good starting point and the full blown profesional CAM package appears to be free but might not be in the end.  Dont forget about post processors,  CAM does not know what machine you have or how to get your machine to move, A post processor is basicalyy a script file which is imported into your cam application to give it your machine instructions (they are free and easly available) say for example you use GRBL (your machines controller/firmware) you would need the GRBL post processor for eithe cambam, fusion whatever.  This is normal even for big machines, if you get a heidenheim controller it could be on a Fanuc or Bridgeport mill.  The person would be looking for their specific machine and controllers post.

I have tried to answer with the information i struggled to bleed out of people when i first started out.  It should be good food for thought.

MeJasonT, 2020-07-14:
A quick search "dxf to gcode" into google gave:-
[https://listoffreeware.com/free-dxf-to-gcode-converter-software-windows/](url)