gnea/grbl #893: Cycle Time Estimates

tfjield, 2020-07-24:
Is there any way to get an accurate cycle time estimate using grbl itself without running the entire toolpath in real time?

I read somewhere that it could be done with grbl_sim, but that also appeared to run in real time.

MeJasonT, 2020-07-24:
Use a cam simulator like camotics it will give you a job duration estimate.  Grbl is a machine controller and not a user interface. Its your machine operating software that would provide this function,  i dont know if candle will do it but as far as i know there isnt a GUI/Program used for Grbl that will.  Im sure Fusion360 (or Confusion360 as i like to call it) will be able to estimate job time in its simulator.

tfjield, 2020-07-24:
I don't know about camotics, but neither Fusion 360 nor Candle provide accurate cycle time estimates that incorporate acceleration.  For a large, beefy machine they're probably reasonably accurate, but for the small, hobby CNC routers where the acceleration may need to be set quite low their estimates are significantly off.  Like by 5X or 10X, depending on the toolpath.

Years ago Autodesk said they were going to incorporate acceleration, but they don't appear to have done it, yet.

I figure the most accurate estimate would come from grbl itself, or a fork of it like grbl_sim.  Since it calculates how many steps to send to the motors, it should be able to calculate how much time it would take.  Perhaps something along the lines of running the toolpath through with a pulse step length of zero, counting the steps, then multiplying by the real step length.  As long as processing time itself is a small factor in the total real time run, I would think this would be pretty accurate.