gnea/grbl #894: CAN'T FLASH WOODPECKER CNC CAMXTOOL V3.4

Don-Girvin, 2020-07-25:
1. Loaded newest version of Arduino 1.8.13
2. Loaded GRBL library in C:\Program Files (x86)\Arduino\libraries\grbl
3. Disconnected all cables from the board except USB and external power supply
4. Verified communications and baudrate for COM3 (I can communicate with the board)
5. Reset board, unplug/replug USB, removed power/restored, started/restarted Arduino IDE and compiled multiple times
6. Edited C:\Program Files (x86)\Arduino\hardware\tools\avr\etc\avrdude.conf to change addresses to try and trick Arduino, by recompiling back and forth with changes multiple times.
 part parent "m328"
    id			= "m328p";
    desc		= "ATmega328P";
#   signature		= 0x1e 0x95 0x0F;
    signature	= 0x0f 0xca 0x07;
    ocdrev              = 1;
;
part parent "m328"
    id			= "m328pb";
    desc		= "ATmega328PB";
#    signature		= 0x1e 0x95 0x16;
    signature	= 0x0f 0xca 0x07;
    ocdrev              = 1;
;
I have attached a "verbose" compile message file and photo of the board.
[Arduino.pdf](https://github.com/gnea/grbl/files/4976022/Arduino.pdf)
[image]

I don't know what else I can do.  Appreciate any help.

PicEngraver, 2020-07-28:
Maybe this will help.  I have not tried it myself yet.
file:///C:/Users/John/Desktop/SainSmart%20Uploading%20Firmware%20on%20a%20SainSmart%20CNC%20Controller.pdf

sanscorp1, 2021-06-23:
Did you ever solve this issue?

PicEngraver, 2021-06-23:
Gave up installed regular Arduino and controller.

remias, 2021-06-24:
Hi,
CAMXTOOL-V3.5 equ CAMXTOOL-V3.5
https://github.com/remias/CAMXTOOL-V3.5-cnc-board-sch
------
The 100n capacitor on the D12 (:: MISO :: Z-limit) wire can cause a problem. By removing this capacitor (see the link below) I was able to update the CAMXTOOL-V3.5 board firmware to GRBL1.1h [using: ICSP pins of board @ USBasp programmer @ 'Extreme Burner'] . 
((After firmware update I soldered-back the 100n capacitor to the D12 wire..))

info link: https://www.reddit.com/r/CNC/comments/e3bzse/schematics_and_firmware_update_cnc3018_and/
------
[CAMXTOOL_V3'5_Sch(rev0).pdf](https://github.com/gnea/grbl/files/6706084/CAMXTOOL_V3.5_Sch.rev0.pdf)

sanscorp1, 2021-06-24:
Thank you! Did you flash stock GRBL or does the board need a custom firmware?

remias, 2021-06-24:
Hi,
download GRBL-CNC-board .HEX file (https://github.com/gnea/grbl/releases    ::  grbl_v1.1h.20190825.hex)
'Extreme Burner' howto: https://www.instructables.com/Hacking-EXtreme-Burner-for-AVR-Atmega-Devices-Prog/
'Extreme Burner' info: https://extremeelectronics.co.in/software/extreme-burner-avr-1-3-beta-test/
download 'Extreme Burner 1.4.3': http://digital-wizard.net/files/extreme_burner_avr_v1.4.3_setup.exe (working successfully :)
USBasp board: I bought it from Aliexpress.
-----
If one does not want to desolder and back-solder the 100n capacitor, then a 1kOhm resistor can be inserted in the wire [this  CAMXTOOL V3.5 and V3.4 cnc-boards go from the 8x2 header to the ICSP header]: then there is never need to desolder and back-solder the 100n capacitor in case of FW upgrade (if anyone is interested this in my comment above, I will make an explanatory diagram of this as well ..)

sanscorp1, 2021-06-24:
Thanks, I will try this.
I have several USB programmers but will have to google them to see if they are fit for the atmega's.

remias, 2021-06-24:
Hi,
more info >> see >> https://github.com/remias/modification_CAMXTOOL-V3.5-or-V3.4--cnc-board-for-FW-update/upload

remias, 2021-06-24:
Hi,
more info >> see >>https://github.com/remias/modification_CAMXTOOL-V3.5-or-V3.4--cnc-board-for-FW-update

beamzer, 2022-11-29:
I had no troubles flashing my board (camxtool v3.5) with grbl 1.1h, just needed to select board ARDUINO Duemilenove Diecimila, no hardware modification nescessary. It doesn't work when you have Arduino Uno selected.

marcelrv, 2024-12-19:
> I had no troubles flashing my board (camxtool v3.5) with grbl 1.1h, just needed to select board ARDUINO Duemilenove Diecimila, no hardware modification nescessary. It doesn't work when you have Arduino Uno selected.

Did the trick for me as well. Thanks!