gnea/grbl #895: Plain GRBL and tool change (beginner question)

JensHauser, 2020-07-25:
Hi,

for some time I´ve been working with a X/Y/Z edge finder like this and it worked

[image]

Now I bought this little thing in order to be able to change a tool during machining a job:
[image]

Seems "G43.1 Dynamic tool length offset" is a subject here. In other forums they use some scripting language to calculate "old" and "new" offset and tell GRBL to use the difference. But with a simple GRBL G code sender with out interaction features?

It´d be nice to

* home X/Y/Z
* use a the aluminum plate to find G54
* start with tool #1
* move to tool change position
* measure (whatever that means) the new tool
* go on with machining

Or am I on the wrong track?

The basic procedure after fixing the work piece would also be helpful.

Thanks in advance

neilferreri, 2020-07-25:
Sorry for the upcoming short response...Check out CNCjs and these [macros](https://github.com/cncjs/CNCjs-Macros) .
I'd avoid G43.1.
Let me know if you have questions... I'll have more time later.

JensHauser, 2020-07-25:
I‘ll have a look, thanks.
In parallel, I‘m still curious to use plain GRBL G code, if possible.

Schildkroet, 2020-07-25:
Take a look at my STM32 Port. I does exactly what you want. After M6 the head moves up and you can change tools. Then it probes the new length (with Tool length sensor) and continues the program.

neilferreri, 2020-07-25:
> I‘m still curious to use plain GRBL G code, if possible.

"Plain" grbl does not handle the M6.

JensHauser, 2020-07-26:
Plain with the exception of the first post.

But independent of that: what is the typical procedure?

MARIOBASZ, 2020-07-26:
I am using bcnc, you can see there how it handles the tool change when receiving M6 (that I know grbl does not recognize M6)

JensHauser, 2020-07-27:
Ok, got it. Thanks a lot to all

closed 2020-07-27 by JensHauser