gnea/grbl #897: target exceeds

Harrycheng1992, 2020-08-01:
Hello everyone

I have some issue like the following picture:
When I tried to control manually then got an error message "jog target exceeds machine travel. command ignored grbl gru"
By the way when I used "Start Homing" the Z axis motor will run then can't stop.
Sometimes it will show alarm 2
[image]
[image]
[image]

It seems to position problem (travel limits?), but no matter how I use the command ''G28'' or the ''home button'' that didn't work.

some details

1. Arduino Mega 2560
2. Ramps V1.4
3. a4988 x5
4. NEMA 17 stepper motor x 5
5. 12V adaptor

I'm beginner needs a lot of help to solve this problem

Good day

otoyasan, 2020-08-01:
こんにちはハリー
物理的なリミットスイッチが無いのにホーミングをオンにしてませんか？写真右下のENDSTOPに何も繋いでないように見えます。
リミットスイッチを使わないなら、ホーミングは出来なかったはず。
$20〜23を全て0に設定して確認する事を提案します。
解決します様に祈ります。

thank you

Harrycheng1992, 2020-08-02:
> こんにちはハリー
> 物理的なリミットスイッチが無いのにホーミングをオンにしてませんか？写真右下のENDSTOPに何も繋いでないように見えます。
> リミットスイッチを使わないなら、ホーミングは出来なかったはず。
> $20〜23を全て0に設定して確認する事を提案します。
> 解決します様に祈ります。
> 
> thank you

Hi otoyassn

It is working!

Thanks a lot

Harrycheng1992, 2020-08-02:
> こんにちはハリー
> 物理的なリミットスイッチが無いのにホーミングをオンにしてませんか？写真右下のENDSTOPに何も繋いでないように見えます。
> リミットスイッチを使わないなら、ホーミングは出来なかったはず。
> $20〜23を全て0に設定して確認する事を提案します。
> 解決します様に祈ります。
> 
> thank you

Another question is : 

Does Z-position in Ramps 1.4 support 2 motors?

When I connect two motors it wasn't run in Z-position but had some noise.

[image]

otoyasan, 2020-08-03:
こんにちはハリー。
Zのモーターが1つなら動くと仮定して話します。A4988に原因があるかもしれません。
このドライバは最大で2Aまで電流が流せると記憶してます。また、流せる電流をポテンショメータで調節します。ここはそれについて話す所ではないので詳しくは省きます。
あなたの繋いでいるモーターが１つ1A近く電流を消費するか、電流調整がなされていて多く流れなくなっているか、などが考えられます。
あなたがA4988に詳しくないなら、メーカーのデータシートなど確認してみるべきだと思います。
また、電源アダプターの出力も気になります。

Harrycheng1992, 2020-08-03:
> こんにちはハリー。
> Zのモーターが1つなら動くと仮定して話します。A4988に原因があるかもしれません。
> このドライバは最大で2Aまで電流が流せると記憶してます。また、流せる電流をポテンショメータで調節します。ここはそれについて話す所ではないので詳しくは省きます。
> あなたの繋いでいるモーターが１つ1A近く電流を消費するか、電流調整がなされていて多く流れなくなっているか、などが考えられます。
> あなたがA4988に詳しくないなら、メーカーのデータシートなど確認してみるべきだと思います。
> また、電源アダプターの出力も気になります。

Hi otoyassan

I set 0.6V to A4988 because each of my stepper motor limited current is 1.8A. 

In my understanding if I want to have 2 stepper motors in Z axis, so I need increase the voltage for two times(1.2V)?

otoyasan, 2020-08-03:
その質問は、例えば
https://github.com/laurb9/StepperDriver
ここで聴いてみるのが適切だと思います。
VREFの計算は、検知抵抗の値が分からないとできないのでは？例えば私のA4988は検知抵抗0.1Ωなので、VREF0.6Vなら電流は0.75Aになるのでは？
私の考えが間違っているかもしれません。

Harrycheng1992, 2020-08-03:
> その質問は、例えば
> https://github.com/laurb9/StepperDriver
> ここで聴いてみるのが適切だと思います。
> VREFの計算は、検知抵抗の値が分からないとできないのでは？例えば私のA4988は検知抵抗0.1Ωなので、VREF0.6Vなら電流は0.75Aになるのでは？
> 私の考えが間違っているかもしれません。
[image]

My a4988 is 0.1Ω.

According to formula "VREF=8⋅IMAX⋅RCS" ：

    Stepper motor rate current: 1.68 A
       
      8*1.68*0.1=1.344    For safety VERF=8*1.5*0.1=1.2

Each axis driver(A4988) needs to set the VERF(V) about 1.2V or Z axis is 2*1.2+2.4V is that correct?

And my adaptor is 
[image]

Thanks

Harrycheng1992, 2020-08-03:
> その質問は、例えば
> https://github.com/laurb9/StepperDriver
> ここで聴いてみるのが適切だと思います。
> VREFの計算は、検知抵抗の値が分からないとできないのでは？例えば私のA4988は検知抵抗0.1Ωなので、VREF0.6Vなら電流は0.75Aになるのでは？
> 私の考えが間違っているかもしれません。

All axis is 32 microstep

Harrycheng1992, 2020-08-03:
> その質問は、例えば
> https://github.com/laurb9/StepperDriver
> ここで聴いてみるのが適切だと思います。
> VREFの計算は、検知抵抗の値が分からないとできないのでは？例えば私のA4988は検知抵抗0.1Ωなので、VREF0.6Vなら電流は0.75Aになるのでは？
> 私の考えが間違っているかもしれません。

I find the actual problem! 

I change my Z axis feed rate so it working greatly with 2 stepper motors(A4988 VERF=1.2V).

Harrycheng1992, 2020-08-04:
> 電源アダプター、出力1Aですか？火事に気をつけてください。
> 私なら少なくとも5Aの物を用意します。
> 
> マイクロステップも、A4988は16までしかなかったと記憶してます。

Ok

I will check all my settings.

You help a a lot. Thank you!