gnea/grbl #901: Machine Crash with Large I, J values.

Rybec, 2020-08-07:
In the following Gcode, lines 1444 and 1446 cause the spindle to head off in a straight line, milling through any material in the way, until it reaches the edge of the machine, where it continues trying to move, straining the motors.  (This happens 20 or 30 times in the full program, and it is always where I and J values are very large.)

1443  G2 X5.754 Y6.148 Z-4.500 I-7.998 J-7.779 K0.000 F150.000
1444  G2 X6.508 Y5.046 Z-4.500 I-2776625.665 J-1898932.711 K0.000 F150.000
1445  G2 X7.324 Y3.220 Z-4.500 I-12.358 J-6.621 K0.000 F150.000
1446  G2 X7.669 Y1.855 Z-4.500 I-1533995.079 J-387777.227 K0.000 F150.000

This was generated with FreeCAD.  I am not sure whether the bug here is with grbl (my machine is a Genmitsu 3018-Pro, running grbl 1.1f) or with FreeCAD's Gcode.  I am aware that the arc length is fairly small and that this might be better as a straight line (and with the massive radius practically _is_ a straight line already), but I want to know exactly what is wrong, so I can report the bug in the right place.  (I think I can resolve the issue in this program, merely by changing those arcs to straight lines, however a bug that can potentially damage a machine is a serious bug that needs attention.)

I have a few questions that might help me to decide whether to file a bug report against FreeCAD:

Is there a maximum magnitude for I and J, in grbl or in Gcode in general?  I searched Google with no luck, but most questions are about maximum _program_ size, so it's possible the information exists but is buried.  My assumption is that it is only limited by the precision of the variable type used internally, which appears to be 32 bit float, which will easily handle the above values.

Is this the result of floating point math error?  If it is, then the issue might be something grbl needs to handle, maybe by setting maximum values for I and J (or a maximum arc to radius ratio), where an error code is emitted instead of driving the spindle off into the distance forever.

davidelang, 2020-08-07:
the arduino only has single precision floating point numbers, so when you give 
it too many digits, it looses precision (IIRC it only keeps about 5 digits), so 
very large arcs don't work.

I don't know the code enought to know where it's crashing (is it in the input 
routine not parsing the number properly, or in the movement planning routines 
that can't handle numbers that large), but for whoever is digging into the code, 
thats a place to start.

David Lang

  On Fri, 7 Aug 2020, Rybec wrote:

> In the following Gcode, lines 1444 and 1446 cause the spindle to head off in a straight line, milling through any material in the way, until it reaches the edge of the machine, where it continues [...]

Rybec, 2020-08-07:
Thank you for this information.  It confirms my suspicions.

If anyone else can verify the specifics that would be awesome.  Specifically, is grbl explicitly truncating, or is the conversion from ASCII to float converting implicitly?  This is important, because 32-bit float has precision ranging from 6 to 9 digits, depending on the specific value.  If it is truncating to exactly 5 digits, then I can write a module to fix the Gcode based on number of digits in I and J, but if it is just fitting the value into float at native float precision, then I can either convert to lines on more than 6 digits, _or_ I can check whether the value will fit 32-bit float precision to maintain the maximum precision grbl can handle.

langwadt, 2020-08-07:
read_float() in https://github.com/gnea/grbl/blob/master/grbl/nuts_bolts.c

Rybec, 2020-08-07:
Thank you so much for that!  That's exactly what I needed.

I have one question, if anyone has an answer:

In the comments for read_float(), there is this line, "For known CNC applications, the typical decimal value is expected to be in the range of E0 to E-4."  Exactly what does this mean?  I presume, given 8 digits of precision, that it means no value should be larger than 99,999,999.0 and no value should be smaller than 0.0001.  Is this correct?