gnea/grbl #904: GRBL on custom hardware

leventelist, 2020-08-10:
I'm fed up with Arduino, and want to design my own controller board. Will GRBL run on the custom board without the bootloader, and other software parts from Arduino?

terjeio, 2020-08-10:
Go 32-bit, there are plenty of options for that available now.

inee-gas, 2020-08-17:
I run one of my routers on the stm blue pill,  never had an issue, the board needs update as it is literally the stm and straight inputs/outputs to the cnc.
Will get around to rebuilding it someday to include opto isolators etc, but as it works well I'm more inclined to leave it.

There is also a port for the esp32 chips (I saw it online but cant remember where) ,it might be worth looking at that option as the esp32 (uk price) is only 7 quid) .there's also a version with a camera for a couple of quid more which you could mount to point at the work surface etc. plus you can run it via usb or wireless, you can even run 2 so one as a controller, and another as maybe a seperate camera or a disply screen using espnow.  But youd have to look it up as i'm new to the esp's