gnea/grbl #905: Arduino stops after some time

leventelist, 2020-08-10:
I'm facing some strange issue. My controller stops after about 29 minutes of operation. The symptom is the same as if I push the HW reset button. I suspected it is some EMC issue, and added a 100nF capacitor to the reset pin of my Arduino, but still no luck. I then linked the GND of my 30V PS and my Arduino. Still no luck.

My next thoughts is the buggy serial interface which is the CH340G. Can this be the issue?

Other is power. Maybe the USB power is not enough for the Arduino, and voltage drops and brown-out detect triggers a reset?

Is there anything else I have to look for?

Also, I've read in the known issues section that the CH340G AND the Original Arduino's serial interface is buggy. Does the FTDI chip also has this bug?

Thanks.

PicEngraver, 2020-08-11:
How are you sending gcode to the Arduino?  Windows laptop computer?  If so, have you set the USB ports to not power down automatically (which is the Windows default setting/action)?

leventelist, 2020-08-11:
I have a desktop computer running Linux and bCNC.

PicEngraver, 2020-08-11:
Sorry, but I can offer no help/advice with either Linux or bCNC.

ASHIT-AXAR, 2020-08-12:
it looks similar to something with the computer "Power Management Settings".
first of all, disable anything related to "power down after ... minutes", "go to sleep after ... minutes", "turn off screen after ..." of your PC.