gnea/grbl #906: Keep position (locking motors) while connecting to GRBL?

Yogsther, 2020-08-10:
When I connect to to GRBL for a second the motors stop trying to keep their position. I'm sure this is a limitation in the software, but it would be great if it was possible to disable that behaviour.

AlexHolden, 2020-08-11:
I'm pretty sure this is caused by the Arduino bootloader. When it first starts up it tristates all the GPIO pins and then waits for a command from the computer. If it doesn't get one it times out and starts running the main program (GRBL in this case). I believe there is a way to use an ISP programmer device to write the firmware directly to the AVR and disable the bootloader.

AlexHolden, 2020-08-11:
I forgot to mention. The reason why the bootloader is running when you connect to GRBL is because opening a new connection to the USB to serial interface causes it to reset the main CPU.

doppelhub, 2020-09-15:
Are you asking about physically plugging in the USB cable (i.e. powering up the microcontroller)?  If so, there's no way you can do this, unless the 5V rail remains present at all times.  It'll still require the modifications described below.

Or are you asking about establishing the software connection multiple times after the USB cable is plugged in?  If so, I'll point out that grbl resets grbl each time a serial connection is established.  This is because grbl's communication protocol must be hand-held by the host... it's a simple "do this" from the host, followed by an "ok" from grbl.  If you fill the buffer, then disconnect and allow grbl to execute the commands, and then later reconnect, you'll have no idea where grbl is (unless it has processed everything and is now idle).  I guess you could choose that implementation (i.e. after connecting, don't send any new commands until the buffer is empty).  

So yeah, all you can handle that limitation, all you need to do is modify when main() runs each xxxx_init() function (i.e. not inside the for loop).  Keep in mind the host will need to be able to understand that grbl might send data to the host immediately upon opening the serial connection.  

One final note: The Arduino IDE's serial monitor automatically resets the serial bus upon connection, so you'll need to use a 3rd party grblController that doesn't reset upon connect.