gnea/grbl #910: Why the machine Homing zero on the (top right) corner?

frviana, 2020-08-19:
I've setup the homing for my machine and was intrigued on why the homing zero on the top right corner. Usually when you X+ or Y+ it goes to the right or top, I would assume the zero point it would be on the lower left corner. Is there a way to change that?

My motor directions are correct so I know this is not the problem.
If I set $23=1 it will zero on the top left which is also not what I believe is correct.

Am I doing something wrong? Is there a way to zero on the lower left corner?

langwadt, 2020-08-19:
top right is just the standard way, which makes sense since it moves everything out of the way.
It doesn't really matter since you would almost always be using work coordinates and not machine coordinates

frviana, 2020-08-19:
@langwadt Thank you for the explanation

closed 2020-08-19 by frviana