gnea/grbl #912: Recommended motor driver setup?

digimbyte, 2020-08-24:
new to Arduino cnc, I loaded the default setup to my ESP32 but don't know what to do to assign pins.
I found one document saying what pins do what
https://github.com/bdring/Grbl_Esp32/wiki/Setting-Up-the-I-O-Pins

but as far as I know my stepper motors require a drive matrix for a ULN2003A

doppelhub, 2020-09-15:
Looks like you're using bipolar stepper motors... those will need to connect to a stepper driver (e.g. DRV-8818)... your microcontroller will then send STEP+DIR pulses to move the steppers.

digimbyte, 2020-09-22:
my steppers came with ULN2003APG's 
I assume I swap this chip/board out for a different driver as this one requires the IN1,IN2,IN3,IN4 matrix
or does another board attach to this as the driver?
I found some DRV8825 boards and I'll see how they go. 
not sure what pins would go to where however.
https://www.aliexpress.com/item/1005001374747956.html

digimbyte, 2020-09-22:
Ultimatly, I feel I need the 6 pack universal CNC controller
https://www.tindie.com/products/33366583/6-pack-universal-cnc-controller/
but I don't have essentially $120 - is it needed? can I just make a basic version myself?
after conversion $70~ + shipping $40 to my country