gnea/grbl #913: G49 triggers an error number 24

sphilton, 2020-08-24:
Combining a G49 gcode (Cancel Tool Length Compensation) with a G0 / G1 / G2 / G3 in the same block produces an error 24 (Two G-code commands that both require the use of the XYZ axis words were detected in the block). This seems strange as a G49 does not use the axes. The issue can be reproduced simply with the following code:
`G0 G49`
`M2`
Is this behaviour intended?

terjeio, 2020-08-24:
from gcode.c:

>// NOTE: The NIST g-code standard vaguely states that when a tool length offset is changed,
// there cannot be any axis motion or coordinate offsets updated. Meaning G43, G43.1, and G49
// all are explicit axis commands, regardless if they require axis words or not.

sphilton, 2020-08-25:
Many thanks for the clarification!

closed 2020-08-25 by sphilton