gnea/grbl #914: ripristino firmware alle impostazioni di base

distebia, 2020-08-25:
Salve a tutti, alimento il mio arduino uno r3 attraverso la porta usb del pc. Su arduino ho caricato grbl 1.1h e ho fatto le varie impostazioni ($30=1, $20=1 ecc...) ma succede che lasciando un pò di tempo il pc spento, e quindi anche arduino non alimentato, all'accensione devo ricaricare una copia di backup delle impostazioni. Mettiamo che arduino è difettoso e non riesce a tenere in memoria se non alimentato il firmware, dovrei avere arduino vuoto ma grbl c'è e risponde ai comandi ma è senza impostazioni e, dopo aver ricaricato il backup, tutto funziona come prima. Come posso risolvere? Grazie

distebia, 2020-08-27:
Possibile che questo problema succede solo a me?

onekk, 2020-09-08:
magari postando in inglese su un forum tutto in inglese potresti ottenere risposte.

Al di la di quello, alimentare arduino da PC non è il miglior modo per non avere problemi.
In più la descrizione del problema non è corredata da nessuna informazione aggiuntiva, ad esempio copie di risposte da terminale, che sono per definizione "internazionali"

Saluti

Carlo D.

distebia, 2020-09-08:
grazie per il consiglio di postare in inglese. non ho risposte dal terminale, quando non lo accendo per un po di tempo, dopo che apro il programma, aver scelto la porta com e aver ricevuto la versione di grbl installata, la prima cosa che faccio è $H, e come risposta ricevo "setting disable" o qualcosa del genere. A questo punto mi rendo conto che non ho i settaggi, richiamo una copia di backup e la ricarico. poi tutto funziona come prima. Circa il consiglio di non alimentare arduino da pc cosa intendi? Fornire tramite jack 7-9V direttamente ad arduino e lasciare il cavo usb inserito? (ho letto che arduino rilevando un'alimentazione esterna disabilita l'alimentazione da usb). Ho cercato anche il modo di fare una sorta di batteria tampone da attaccare ad arduino ma non ho trovato niente.

onekk, 2020-09-10:
Verifica che siano attive alcune voci nella configurazione, oppure hai un problema di eeprom che non è attivata per qualche motivo, magari per impostazioni dell'arduino ide.

A  memoria non ricordo, avendo settato le cose molto tempo fa, mi pare che però alcune cose vadano settate in fase di compilazione.

Controllo e ti faccio sapere.

Saluti

Carlo D.

distebia, 2020-09-10:
su arduino prima ho inviato clear eeprom, poi blink, successivamente ho caricato grbl 1.1h

onekk, 2020-09-11:
Sorry for the italian thread,

Summary of the thread:

EEPROM problems, Arduin don't retain values and forgot it between power cycles.

The following is the prosecution of the discussion in Italian language because one partecipants apparently don't know enough english to explain the problem.

--
Nota doverosa per il forum inglese, magari qualcuno ci potrà suggerire soluzioni migliori.

Hai seguito questo processo, che è quello raccomandato?

https://github.com/gnea/grbl/wiki/Compiling-Grbl

Ci sono delle cautele da seguire, sembra che consiglino di flashare prima uno sketch, non vorrei che quello sketch imposti alcune variabili di FUSE (che sono alcune preset del processore) che poi influenzino il vero firmware.

Ci sono dei comandi per resettare la eeprom e riscriverla, "RST= " e poi qualcosa a seconda dell'operazione, se GRBL trova la eeprom sporca allora la riscrive automaticamente, vedi nel codice di config.h. attorno alla riga 500.

Spero che aiuti.

Saluti

Carlo D.

distebia, 2020-09-12:
di solito faccio prima eeprom-clear e poi carico l'esempio basic-blink, poi carico grbl. ho guardato il config.h e trovo abilitate (non commentate) le righe di "RST=" attorno alla riga 500 e le lascio come sono.
# define  ENABLE_RESTORE_EEPROM_WIPE_ALL          // '$ RST = *' Default abilitato. Commento da disabilitare.
# define  ENABLE_RESTORE_EEPROM_DEFAULT_SETTINGS  // '$ RST = $' Default abilitato. Commento da disabilitare.
# define  ENABLE_RESTORE_EEPROM_CLEAR_PARAMETERS  // '$ RST = #' Default abilitato. Commento da disabilitare.

devo fare altro?

onekk, 2020-09-15:
devi mandare i comandi per pulire la eeprom:

 $ RST=*

e poi 
$ RST=$ 

per impostare i parametri di default.

poi riconfigurare GRBL, se è un problema di eeprom sporca, in questo modo la pulisci e la inizializzi.

Se non va a questo punto è un problema di eeprom.

Nella discussione mi pare dicano di non caricare nessuno sketch, vedi il link che ho postato in precedenza, e di partire con un Arduino IDE "pulito", infatti GRBL usa il C e non il C++ e quindi lo sketch GrblUpload.ino semplicemente carica  grbl.h.

Saluti

Carlo D.

closed 2023-09-09 by distebia