gnea/grbl #915: Spiral Ramping G-code file runs much slower on curves

baconWood, 2020-08-25:
The attached G-code file of a spiral ramp runs much slower on curves on my Arduino Mega.

I'm running with the following parameters:
$100=62.992	X-axis steps per millimeter
$101=62.992	Y-axis steps per millimeter
$102=125.984	Z-axis steps per millimeter
$110=5080	maximum rate, mm/min X

$111=5080	maximum rate, mm/min Y
$112=2540	maximum rate, mm/min Z
$120=90	acceleration, mm/sec^2 X
$121=90	acceleration, mm/sec^2 Y
$122=90	acceleration, mm/sec^2 Z

Any help or suggestions appreciated.  Thank you!

[SprialRamp.txt](https://github.com/gnea/grbl/files/5126099/SprialRamp.txt)

MARIOBASZ, 2020-08-26:
Increase aceleración?
Im using maximum speed aproximately  4000 and acceleration 250 or more

doppelhub, 2020-09-15:
Is it stuttering (increase baud rate, or disable "convert arcs to line segments" in your CAM generator)?
Or are you running into your acceleration limit (increase $120/121/122)?