gnea/grbl #916: Exclamation point "!" in comment puts machine into Hold state.

TheRealDJ, 2020-08-26:
When your GCODE comment contains an exclamation point "!", the machine goes into Hold state. I know that "!" is the realtime command for pause, but in a GCODE comment, I think it should be ignored.

Took many hours to troubleshoot what was causing the hold state.

Example failing GCODE ... I start with my machine @ Z=15.000. When I run the below, I get a Hold at about 15.035:
```G94 ( Millimeters per minute feed rate. )
G21 ( Units == Millimeters. )
G91.1 ( Incremental arc distance mode. )
G90 ( Absolute coordinates. )
G00 S10000 ( RPM spindle speed. )
G00 F1000 Z25.0000 ( Retract )
G04 P1.0000
( DJ says hello! )
G00 Z10.0000 ( Lower )
```

Example working GCODE (just removed the "!") ... Machine moves to Z=25, waits 1 second, then moves to Z=10:
````
G94 ( Millimeters per minute feed rate. )
G21 ( Units == Millimeters. )
G91.1 ( Incremental arc distance mode. )
G90 ( Absolute coordinates. )
G00 S10000 ( RPM spindle speed. )
G00 F1000 Z25.0000 ( Retract )
G04 P1.0000
( DJ says hello )
G00 Z10.0000 ( Lower )
```

Thank you.
DJ

neilferreri, 2020-08-26:
What sender are you using?

langwadt, 2020-08-26:
doesn't most senders strip out comments?, sending them is just a waste of bandwidth and processing power

TheRealDJ, 2020-08-26:
I am using CNC.js. I understand the comment, but this shouldn't stop the machine. I figured this had not been encountered before. If its not a concern, no big deal. I removed the "!" from my comment ... My actual code said ( Retract bit just to be safe! )

neilferreri, 2020-08-27:
I've seen the same with CNCjs...maybe bring it up there?

doppelhub, 2020-09-15:
This is due to how grbl processes runtime commands (!~?, etc)... the serial ISR picks the character off immediately... it never enters the buffer, but instead is processed immediately.  I agree this should be documented, but the behavior is expected.  Take a peak at serial.c for more info on how the serial stream is processed on its way to system_execute_line() & gc_execute_line().