gnea/grbl #920: Spindle refuses to spin after limit switches installed

GuillaumeRives, 2020-09-11:
I've a CNC with a WoodPecker 3.4 board, I've installed all limit switches (2 by axis), I've put X limit switches to Z limit labelled pins like other people advised. My homing is working great but my spindle refuses to work if my Z limit switches are connected (to X limit labels).

What can I do ?

PicEngraver, 2020-09-12:
Can you post a picture of your switch connections?

GuillaumeRives, 2020-09-12:
Here you go sir. Another info, my switches are NO. Do I really need NC switches in order to avoid any noise ? switches are already packed with resistors and capacitors (well known red pcb packages for reprap projects).
[image]

PicEngraver, 2020-09-12:
Well. I have the same setup, except I have the small version of your machine that I use a a software test machine only.  I have not used the spindle on it.  We also probably have the same limit switches.  I do not use hard limits at all.  They have been too much trouble on my other machine so I only use soft limits.  On my test machine the switches are used for homing only.

From your description of your problem, I suspect that you have a bad board.  Hopefully you are still under warranty and can get it replaced.  There is nothing else I can think of that will cause your issue.

Just to be sure, does it happen when when you send a manual command like 'G1 X0 M03 S1000?' or only when running a gcode file?

GuillaumeRives, 2020-09-12:
I didn't try to send a gcode file because I was afraid of destroying the bits. Whenever I try to send a M3 S1000 or clicking the Spindle Start / Stop button, the spindle has this behavior. I must admit that hard limits are pretty much useless when you know what you do... I had those remaining switches so I installed them. I tried to install only homing switches but it's the same. As soon as my Z switches are installed, the spindle refuses to spin.

PicEngraver, 2020-09-12:
Does it do the same if the bottom Z limit switch is removed/disconnected?  It is pretty much useless in most cases anyway, IMO.  Since homing works OK, the top Z switch has to be OK.

PicEngraver, 2020-09-12:
Try sending a file with the bits removed - just do a air-cut.

thomas374b, 2020-09-18:
Are you aware that Spindle-Enable-Bit and Z-Limit-Bit are swapped if you configure VARIABLE_SPINDLE ?
This is due to the need of hardware-PWM capable bit.  If you have VARIABLE_SPINDLE defined keep sure that your 
wiring is correct.

GuillaumeRives, 2020-09-18:
Sorry for the lack of answers.. I didn't get time to try out your different solutions since I had a surgery. I still read your comments though.
@thomas374b : I wasn't aware of that problem. I have some questions then : 
1. How am I supposed to wire my z limit switches and my spindle then ? Do I needto swap spindle and z limit switches ?
2. If I comment VARIABLE_SPINDLE, I suppose that my spindle will only run at full speed and my problem will be resolved right ?

PicEngraver, 2020-09-18:
You do not have to change your wiring, assuming it is all wired as you show.  The board you have takes care of all pin assignment correctly.  I have the same board and it works correctly.  I strongly suspect you have a defective board from the way your system acts.  There is a possibility you could have a shorted connection in your wiring that is not shown in your photo, but I assume you have checked that.

GuillaumeRives, 2020-09-18:
@PicEngraver I checked all connections twice and I don't believe I have a shorted one. One thing that might be a source of problems is that I upgraded the firmware with the latest version of GRBL. Could this be the problem ? Maybe the default settings of GRBL aren't the same as the default settings of the firmware originally uploaded on the board ?

PicEngraver, 2020-09-18:
I just installed my spindle for the first time and connected it to the board with supplied cable, and it works fine.

PicEngraver, 2020-09-18:
I am using grbl version 1.1f.  These are my current settings:
$0=30 (step pulse, usec)
$1=254 (step idle delay, msec)
$2=0 (step port invert mask:xxxxxxxx)
$3=2 (dir port invert mask:xxxxxxxx)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=0 (status report mask:xxxxxxxx)
$11=0.005 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=1 (report inches, bool)
$20=1 (soft limits, bool)
$21=0 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=3 (homing dir invert mask:xxxxxxxx)
$24=50.000 (homing feed, mm/min)
$25=2000.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$30=1001 (rpm max)
$31=0 (rpm min)
$32=0 (laser mode, bool)
$100=810.000 (x, step/mm)
$101=810.000 (y, step/mm)
$102=810.000 (z, step/mm)
$110=2000.000 (x max rate, mm/min)
$111=2000.000 (y max rate, mm/min)
$112=2000.000 (z max rate, mm/min)
$120=50.000 (x accel, mm/sec^2)
$121=50.000 (y accel, mm/sec^2)
$122=50.000 (z accel, mm/sec^2)
$130=171.450 (x max travel, mm)
$131=85.000 (y max travel, mm)
$132=40.000 (z max travel, mm)

GuillaumeRives, 2020-09-18:
Ok so... I played with my settings a little bit. You seem to use soft limits while I wired all limit switches and intended to use hard limits. This is my problem. As soon as I disable hard limits and enable soft limits, it works properly... I don't know why though...

PicEngraver, 2020-09-18:
It sounds like you may have a hard limit trigger alarm condition when your spindle starts.  That may be from electrical interference (EMI) issues.  Whenever a hard limit triggers, it will shut off the spindle, at least it does with my setup.  You might try physically separating your Z limit switch wiring from your spindle motor wiring as far apart as possible.  Also, try adding a small ceramic capacitor (0.1UF ?) from each spindle motor connection to a ground connection.

GuillaumeRives, 2020-09-18:
I already tried to isolate spindle cables from limit switches cables and it did the same... I dont really want to mess with capacitors and stuff on my machine... I didn't examine this board but in my opinion, this filter should be installed from manufacture... 

Well you know what ? I've just tweaked all my soft limits... It's even better than hard limits since you don't have to reset your cnc when you hit the soft limits... I will stay like this... I don't find any benefit to have hard limits actually... What do you think about that ?

My current settings : 

$0=10
$1=25
$2=0
$3=2
$4=0
$5=0
$6=0
$10=3
$11=0.010
$12=0.002
$13=0
$20=1
$21=0
$22=1
$23=3
$24=50.000
$25=2000.000
$26=5
$27=1.000
$30=1000
$31=0
$32=0
$100=798.000
$101=800.000
$102=800.000
$110=2000.000
$111=2000.000
$112=2000.000
$120=50.000
$121=50.000
$122=50.000
$130=298.000
$131=177.000
$132=42.000

PicEngraver, 2020-09-18:
I can not use hard limits on any of my grbl machines as they are too erratic and subject to many false trips, no matter what conditioning circuitry I have tried.  Soft limits are not perfect, but better than hard limit problems IMO.

GuillaumeRives, 2020-09-18:
Yeah I can see that... I'm new to GRBL and CNCs (A lot of 3D printing expérience though and never saw any hard limit on any 3D printer and it works perfectly). I take note of your experience, It relieves me because this problem was driving me mad...

Thanks for your help ! I close this issue 👍

closed 2020-09-18 by GuillaumeRives