gnea/grbl #921: grbl thinks Limit switch is pressed when running steppers.

Schwarzi07, 2020-09-13:
My Problem is that every time i move the stepper for x or y i get an Alarm saying taht a limit switch is being pressed.
The Alarm activates when i move the steppers via jog or with GCode, but it doesn´t activate when i use the Home Funhktion.
I tried shielding the wires and twisting them, but it didn´t help.

Here is my grbl Config:

$0=10 (Step pulse time)
$1=25 (Step idle delay)
$2=0 (Step pulse invert)
$3=1 (Step direction invert)
$4=0 (Invert step enable pin)
$5=0 (Invert limit pins)
$6=0 (Invert probe pin)
$10=111 (Status report options)
$11=0.010 (Junction deviation)
$12=0.002 (Arc tolerance)
$13=0 (Report in inches)
$20=0 (Soft limits enable)
$21=1 (Hard limits enable)
$22=1 (Homing cycle enable)
$23=11 (Homing direction invert)
$24=100.000 (Homing locate feed rate)
$25=600.000 (Homing search seek rate)
$26=250 (Homing switch debounce delay)
$27=1.000 (Homing switch pull-off distance)
$30=1000 (Maximum spindle speed)
$31=0 (Minimum spindle speed)
$32=1 (Laser-mode enable)
$100=133.000 (X-axis travel resolution)
$101=133.000 (Y-axis travel resolution)
$102=133.000 (Z-axis travel resolution)
$110=900.000 (X-axis maximum rate)
$111=900.000 (Y-axis maximum rate)
$112=900.000 (Z-axis maximum rate)
$120=50.000 (X-axis acceleration)
$121=50.000 (Y-axis acceleration)
$122=50.000 (Z-axis acceleration)
$130=450.000 (X-axis maximum travel)
$131=500.000 (Y-axis maximum travel)
$132=200.000 (Z-axis maximum travel)

Does anyone have an Idea why this is happening?

PicEngraver, 2020-09-13:
It is probably inherent in the design of the Arduino Uno circuitry.  I have yet to find a completely successful hardware solution, so I use and recommend the use of soft limits ($20=1) and leave hard limits disabled ($21=0).

peter76, 2020-09-14:
Did you use pull-up resitors and decoupling capacitors as explained here:

https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches

I had the same problem initially, but adding the pull-ups and decoupling capacitors solved it.

Regards, Peter

doppelhub, 2020-09-15:
OP: this is a noise issue.  You need to add a hardware lowpass filter.

> I have yet to find a completely successful hardware solution, so I use and recommend the use of soft limits ($20=1) and leave hard limits disabled ($21=0).

The GrBLDC Uno shield has zero issues with limit switch tripping.  Here's the schematic:
https://github.com/Defdist/GrBLDC3/blob/master/Schematic%20-%20RevB%20-%20BOM2.pdf

Schwarzi07, 2020-10-06:
I used an opto-couppler and it works perfectly.

closed 2020-10-06 by Schwarzi07