gnea/grbl #922: Problems with updating to the newest version

ruma995, 2020-09-17:
Hello guys,
I'm trying to flash the v0.9 with the latest version, and it pop out this error:
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x0d
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x0a
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x47
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x72
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x62
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x6c
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x2e
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x39
An error occurred while uploading the sketch

Any ideas how this can be done?
Thank you!
Mario

thomas374b, 2020-09-18:
Version 0.9 is from 2016. Have you tested with recent version ? We have 1.1h from Oct 2019.

ruma995, 2020-09-18:
On controller was version 0.9 and now I'm trying to update it with 1.1h, and this errors occured while uploading the sketch

doppelhub, 2020-10-02:
Try unplugging the shield.  If you can now program, then your arduino shield is preventing the 16U2 from pulling the 328p's reset line low.

LongLiveCHIEF, 2020-10-03:
Here's what I ran to get mine to work (was getting the same errors you were prior switching my flashing process to the command below):

```
sudo avrdude -p m328p -b 57600 -P /dev/ttyUSB0 -c stk500v1 -U flash:w:grbl.hex
```

ruma995, 2020-10-05:
Hey Brian, 
can You please help me how I ran this, or what to do with it. I tried to copy it but doesn't work. Sorry but I have very little knowledge about Arduino. Thank you for your help.

* I tried with unplugged shiled but the same errors pops out