gnea/grbl #930: Need Help with Homing

wollo-fpv, 2020-09-23:
Hello,

I want to setup my small cnc pcb with homing
On x and y axis I have installed the end stop on the minus side.
On the z axis I have mounted the switch on the top, plus side.
I have only 1 switch on each axis.
The grbl controller is a,

MKS DLC V2

All direction of the axis are configured well.
I try to get the Z axis working.

**** Connected to ttyUSB0 @ 115200 baud ****
Grbl 1.1h ['$' for help]

$5 = 1    (Invert limit pins, boolean)
$20 = 1    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 4    (Homing direction invert, mask)
$132 = 75.000    (Z-axis maximum travel, millimeters) (the Z axis have a travel of 80mm. I just put it on 75mm to be safe)

Testing end stop on Z axis
I power up the the machine.
When I push the switch with the finger, the LED on the switch is lighting up, GRBL report a error.
In this case it is right!

“An unexpected error was detected: (ALARM:1) Hard limit has been triggered. Machine position is likely lost due to sudden halt. Re-homing is highly recommended.”

I guess that my switch is connected and is working well.

As soon I send  $H,
Z axis is moving up. 
The LED on the end stop is lighting up.
But the Z axis don´t stop to move up 
GRBL is ignoring that that the switch is triggered 

I flashed GRBL with the .HEX from Github.
No changes made in firmware.

Can some one help please…..

Kind regards,

Wolf
 

$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 2    (Step pulse invert, mask)
$3 = 2    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 16    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 1    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 4    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 100.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 15.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 400.000    (X-axis travel resolution, step/mm)
$101 = 400.000    (Y-axis travel resolution, step/mm)
$102 = 400.000    (Z-axis travel resolution, step/mm)
$110 = 1500.000    (X-axis maximum rate, mm/min)
$111 = 1500.000    (Y-axis maximum rate, mm/min)
$112 = 400.000    (Z-axis maximum rate, mm/min)
$120 = 30.000    (X-axis acceleration, mm/sec^2)
$121 = 30.000    (Y-axis acceleration, mm/sec^2)
$122 = 50.000    (Z-axis acceleration, mm/sec^2)
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 75.000    (Z-axis maximum travel, millimeters)
ok

helLf1nGer, 2020-09-23:
Does it trigger after the switch is released, or during? I had a similar issue when I connected 5v instead of a signal cable to grbl

wollo-fpv, 2020-09-24:
Problem solved!
As well I cant understand well!
$5 = 1 (Invert limit pins, boolean)
must be 
$5 = 0 (Invert limit pins, boolean)
And 
$23 = 4 (Homing direction invert, mask)
must be
$23 = 3 (Homing direction invert, mask)
Homing working well....!
Thx

closed 2020-09-24 by wollo-fpv