gnea/grbl #931: adding higher level GRBL interfaces to the wiki

naikymen, 2020-09-24:
I believe it would be great to hold a list of projects that support GRBL in the wiki.

At first I started working on one from scratch, perhaps due to my finite ability to google.

Though I have learnt much, the curve was very steep, and I would have preferred using an existing project.

I hope you find this agreeable.

My incomplete suggestion for the wiki follows. I have only included open-source projects.

```
* [Higher level interfaces](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#higher-level-interfaces)
```

```
-----

# Higher level interfaces

With the information above you should be able to write or modify an interface to GRBL.

A non-exhaustive list of projects follows.

## GUI applications supporting GRBL

* [bCNC](https://github.com/vlachoudis/bCNC) (open source, python based)
* [UGS](https://winder.github.io/ugs_website/) (open source, java based)

## Python

* [gerbil](https://github.com/michaelfranzl/gerbil) (open source, python3 module)
```

naikymen, 2020-09-24:
Perhaps some LMGTFY links wouldn't be amiss :)

terjeio, 2020-09-24:
[My sender](https://github.com/terjeio/Grbl-GCode-Sender) is in the final stages of beta testing, written in C# (for Windows only) using WPF and VMMV coding patterns. It has a layered architecture, components may be used to build a different UI on top of the core if one wish. It should appeal to users of the now discontinued [GrblPanel](https://github.com/gerritv/Grbl-Panel).

breiler, 2020-10-03:
@terjeio that looks really nice, great work!

@naikymen there are some senders specified here:
https://github.com/gnea/grbl/wiki/Using-Grbl#how-to-stream-g-code-programs-to-grbl

In UGS we've tried to create a list of the "competition", @terjeio you just made the list 😁 
https://github.com/winder/Universal-G-Code-Sender/wiki/Software-comparison#gcode-sender

naikymen, 2020-10-03:
That's great @breiler, I don't know how I missed that. Thanks :)

I just added a link to that wiki page to https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface

Perhaps i can add a non-GUI interface tools section around [here](https://github.com/gnea/grbl/wiki/Using-Grbl#python-streaming-scripts-officially-supported-by-grbl-cross-platform). And/Or reorganize it a bit into GUI and non-GUI sections, if it sounds right.