gnea/grbl #932: FR: Ability to tell grbl that homing limit switches are not at machine zero?

positron96, 2020-09-24:
Hello! 
I've installed limit switches at max coordinates of the machine for homing. Now after homing my machine coordinates are zero (actually -1,-1,-1 because of 1mm pulloff). 
But now every time I move to machine zero on any of the axis, my limit switches trip and grbl goes into alarm, requiring to reset!
Is it supposed to work this way? Are there any parameters that can change the position of homing limit switches to be not at zero? If not, than consider this issue a feature request!

This seems important, because some popular programs assume that machine zero is reachable and "safe":
* Candle goes to G53 G0 Z0 as a "safe position"
* Autodesk Fusion 360's grbl post-processor goes to machine 0,0 at the end of the job

langwadt, 2020-09-24:
build with HOMING_FORCE_SET_ORIGIN to set zero after pull off. 
afair fusion doesn't goto 0,0 it does a G28 which is default 0,0. save your preferred safe position with g28.1

positron96, 2020-09-24:
Yeah, Fusion does a G28 if it's enabled, otherwise it goes to machine zero. Surprisingly, I can't find a way to leave the spindle in place at the end.

Anyway, thank you for a hint about rebuilding, I'll try that.

closed 2020-09-24 by positron96

langwadt, 2020-09-24:
don't know if it still there but the fusion post-processor used to have an option to disable "safe retracts" removing the G28, just keep in mind the it also removes the retract at the beginning so it'll G0 directly to starting point. 
I think the better way is to move to a safe position and store that with a G28.1, you'll only have to do it once

neilferreri, 2020-09-24:
The post you're using likely goes to WORK zero.

langwadt, 2020-09-24:
I doubt it, going to work zero would be bad it'll often be somewhere on the part

neilferreri, 2020-09-25:
I should've specified XY work zero at whatever the clearance height is.  As I think about it, an older grbl post used to just retract to the clearance height and hold X&Y.  I just looked at the newest grbl post processor and it moves to machine home.  This position / offset can be changed if you set up a machine profile in Fusion.
I have been using my own, [modified post processor](https://a360.co/2PPwJrT) for several years.  I have it return to a G53 Z-5mm position, with several configurable options.