gnea/grbl #933: Changing feedrate while jogging

javspace, 2020-09-25:
Is it possible to change the speed at which the machine is moving while in jog?

terjeio, 2020-09-25:
I do not think so:

https://github.com/gnea/grbl/blob/eefe2bb95bb7b21ec2bb87e6ab6e20747e1626c4/grbl/jog.c#L24-L31

doppelhub, 2020-10-02:
Instead of sending one jog command, send a bunch of very small jog move requests (with desired feedrate in each command).  Another option is to send just one jog command, and then - if the feedrate changes - cancel the jog command and send a new one.

breiler, 2020-10-03:
What @doppelhub is suggesting works pretty good, we use this in UGS for gamepad/joystick jogging.

If you don't mind digging into Java code here is an example of how we are doing it. 
https://github.com/winder/Universal-G-Code-Sender/blob/master/ugs-core/src/com/willwinder/universalgcodesender/utils/ContinuousJogWorker.java#L116