gnea/grbl #938: Gritty Sound from Steppers

hires, 2020-10-09:
I just got a Sainsmart 3018-PROver CNC running grbl version 1.1f. The machine seems to work but I hear a regular clicking noise from all three steppers. It's a fairly loud pulsing at around 18Hz which seems constant no matter what speed the steppers are running at.

Is there some kind of timing issue or setting that is causing this?

PicEngraver, 2020-10-09:
Check the anti-backlash nuts for looseness.

hires, 2020-10-10:
Thanks for replying. It's definitely an electrical / code problem and not the lead screws. It feels like the code is somehow "missing" the output on a repeated basis. The motors run through normal acceleration curves, but superimposed on all movement is this low speed clicking sound.