gnea/grbl #939: grbl1.1H error 6

prak1111, 2020-10-09:
Hello,
Homing the arduino uno cnc throwing error 6. It moves z axis few steps and throws error 6. all limit switches are wired correctly and are verified using bcnc setup wizard. Here is my configuration but I don't see my configuration using above 30KHz(as it throwing error 6). 

$$ settings-
Grbl 1.1h ['$' for help]
[MSG:'$H'|'$X' to unlock]
$0=10
$1=25
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=3
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=0
$100=2560.000
$101=2000.000
$102=320.000
$110=200.000
$111=200.000
$112=50.000
$120=10.000
$121=10.000
$122=2.000
$130=950.000
$131=1800.000
$132=150.000
ok
$I Output-
[VER:1.1h.20190825:]
[OPT:VH,15,128]
ok

PicEngraver, 2020-10-09:
You have $3=0.  Must be 4 or greater.
Error No. 6 - Value <3usec - Minimum step pulse time must be greater than 3usec.

terjeio, 2020-10-10:
Error 6 or alarm 6? Error 6 is raised when trying to set $0 < 3, alarm 6 is when reset input is asserted during homing.

prak1111, 2020-10-12:
> Error 6 or alarm 6? Error 6 is raised when trying to set $0 < 3, alarm 6 is when reset input is asserted during homing.

Is that mean somehow reset button is trigerred while homing? Actually I am runing grbl1.1h on cnc shield V3 and willing to use PWM for spindle.
So what should i do for reset issue?

PicEngraver, 2020-10-12:
Alarm 6 might mean you have your homing switches wired wrong, or maybe $5=0 set wrong.  Have you tried it set to $5=1?

terjeio, 2020-10-12:
> Is that mean somehow reset button is trigerred while homing?

It is the grbl reset pin that is triggered (labeled as `Abort` on the [shield](https://wiki.keyestudio.com/Ks0160_keyestudio_CNC_Shield_V3)?), not the MCU reset pin (labeled as `E-STOP`?):

`#define CONTROL_RESET_BIT         0  // Uno Analog Pin 0`

If this is not connected to a switch try to connect it to 5V (or GND dependent on your config) since leaving it open may make it sensitive to noise.