gnea/grbl #942: Missing X real time status report per second

amoineau, 2020-10-21:
Hello,

Thank you for your excellent work on this project. 
I'm using GRBL paired with GRBL-plotter and I often get a message saying that we are missing real time status report. I wanted to know how badly does this affect the system and what could cause this problem ?
For information I am not using an arduino board but one designed by a co-worker. Apart from this, everything seems to be working just fine.

Thank you for your help.

svenhb, 2020-10-22:
Which version are you using? In an older version was bug in character-counting which could mess-up the controller...
But now,I don't think this is really a problem. Note: I figured out, during homing no response happens, so I suppress the message during homing.
Perhaps your computer is a bit overwhelmed - did you try with slower poll frequency for the status-report?
[image]

amoineau, 2020-10-23:
Hi,
I'm using version 1.1 of grbl and v1.5.0.5 of GRBL-plotter. I'll try to lower the frequency as soon as I can and get back to you.
I'm note sure I understand what you mean by "I suppress the message during homing".
Thank you for your help, it's highly appreciated.
Regards,
Antoine.

svenhb, 2020-10-23:
>I'm note sure I understand what you mean by "I suppress the message during homing".  
  
During Homing I stop counting the missing reports.
Yesterday I released 1.5.2.1, perhaps problem is solved then.
Regards
Sven

amoineau, 2020-10-26:
Ok you meant "in a new release", I thought so but I wasn't sure. I can't access the machine right now but I will test both fixes as soon as I can. Thank you for being so reactive.

amoineau, 2020-11-12:
Hello,
Sorry for the long delay. I tried using the new release but I can't launch the source code. I downloaded the zip, extracted, opened the solution and pressed F5. I didn't alter anything and I got the following result :

[image]

amoineau, 2020-11-12:
Damn I forgot my VS was in french on this computer... Do you want the error in english ?

amoineau, 2020-11-12:
[image]

svenhb, 2020-11-12:
I forgot to upload the settings files, check here: https://github.com/svenhb/GRBL-Plotter/tree/master/GRBL-Plotter/Properties
Better open an issue at GRBL-Plotter: https://github.com/svenhb/GRBL-Plotter/issues

amoineau, 2020-11-12:
Ok I opened an issue on GRBL-Plotter. What file should I add from  https://github.com/svenhb/GRBL-Plotter/tree/master/GRBL-Plotter/Properties ?

svenhb, 2020-11-12:
Just the the files from 9 days ago

amoineau, 2020-11-12:
Ok thanks !

amoineau, 2020-11-12:
I was able to play around with the machine and the latest release of GRBL-plotter and I didn't get a single "Missing real time status report". However, it will take more time to be sure that the problem is solved. In the meantime, do you want to close the issue ?

amoineau, 2020-11-18:
Ok, the problem did not occur since the update so I guess problem's solved.

closed 2020-11-18 by amoineau