gnea/grbl #944: Maximum Spindle Acceleration Setting

jiquera, 2020-10-22:
This has probably be requested before but would be possible to add a spindle maximum acceleration setting just like with the axis motors? It would make life so much easier on managing current draw from spindles and make much more spindles compatible with specific setups.

I realize I can do this manually by sending a gcode sequence but this seems like a setting that would be useful for a lot of people and doesn't seem particularly hard to implement (at least at first glance).

Cheers

jiq

marvin-yorke, 2020-10-23:
I'd guess that if you're using a VFD controlled spindle, VFD itself would handle spin up and spin down speed. 

I'm using a DC motor with no RPM feedback, so I'm thinking about adding an intermediate circuit with another Arduino-like MCU probably, that will read PWM speed input from GRBL, sense spindle RPMs and control spindle power supply to adjust between the two. I haven't thought about spindle acceleration though, but it might be a good idea, since the motor supplier advises against starting it at full speed, probably because of this very reason. Let me know if you'd be interested in this solution

jiquera, 2020-10-23:
I got a bit confused with the grbl main page so I posted it twice (old and new wiki): https://github.com/grbl/grbl/issues/1677

Of course that would be a pretty neat solution, but I'm also looking for one-size-fits-most solution which could help a lot of cnc beginners that already have a hardware setup. Also, since the goal here is current limiting, I think the accuracy doesn't need to be great. Which is why I think this can be solved in software without a feedback loop.