gnea/grbl #948: Custom firmware additional $ setting.

StuartB4, 2020-11-01:
Didn't really know where to ask this.

On Ortur Laser engravers they use a customised version of GRBL.
They add a G-Shock sensor to the machine so it stops and switches 
the laser off when it's knocked or lifted while it's engraving. 

They've added $33 and set the default threshold setting to $33 = 250.
What I'd like to know is what would the 250 refer to.

Is it a small unit of G force?
Is it a voltage threshold of say 250 millivolts generated by the movement
and sensed by the sensor? or something else.

Any help would be appreciated. 

[image]

terjeio, 2020-11-01:
Isn't it documented in the [manual](https://lasergrbl.com/ortur-manuals/)?

StuartB4, 2020-11-02:
No. No mention of it, they don't go that far in to grbl configuration. It just mentions the fact that it has the sensor and what it does if the machine is move while it's working.

arkypita, 2020-12-18:
Yes, it is the g-shock threshold.

I have talked a little with Ortur and they say that this value it should not be changed by users because it refers to a security feature (g-shock, exactly). It was intentionally left undocumented by Ortur, and all the more reason those of gnea / grbl know nothing about it. In LaserGRBL i have documented it as "G-Sensor threshold".

I have no more information on this (unit of measurement or physical meaning) but it must be left as is by manufacturer's assertion.