gnea/grbl #950: Z axis homing problem

ssombra, 2020-11-02:
I have a DIY laser engraver which I built using an old 3D printer frame and parts, a UNO with a CNC Shield and a 5.5W dolió de láser. I flashed it with GRBL 1.1 and I am using LaserGrbl. Once I had it working, I added endstops switches for X and Y, only 1 per axis, as min switches with the left bottom side as origin (3D printing background :)) and changed the homing sequence for X and Y only, again everything was working good. I then decided to also add an endstops switch for min Z, as my setup included also Z axis motors and I think that it would be a good idea to be able to home Z at the height with the laser is focused on the bed and then be able to move up depending on the height of the material to engrave. 
I installed the switch, connecting it to “Z min”, and changed the homing sequence to be: Z, then X and Y. 
The problem I have is that when I press the home button, the z axis starts moving but it doesn’t stop even when the switch is pressed. I checked for continuity and the switch is working good (I.e, continuity when pressed). Am I missing something? My laser PWM is connected to Z max, does that have something to do? Should I changed something?

JackyLau, 2020-11-10:
Which controller board you are connected ?

ssombra, 2020-11-11:
> Which controller board you are connected ?

Arduino UNO + CNC Shield

JackyLau, 2020-11-15:
The Z limit switch do NOT connect to Z pin (D11), connect to D12 is right  (locate near power connector)

ssombra, 2020-11-24:
> The Z limit switch do NOT connect to Z pin (D11), connect to Spindle Enable pin (D12) is right (locate at the side of “end stop X- pin” or near power connector)

Thanks, I will try it there.

housspin, 2020-11-27:
Hi guys. I have a sainsmart board 1.1f firmware on it (grbl)  and limit
switches on z y x axis. I connected all plugs and every time the switches
are pressed down the board doesn't respond any command. Any idea?
Thanks

housspin, 2020-11-27:
And i have to turn the board off and on to get back to wor