gnea/grbl #952: grbl Upload to Adruino issue

jacooos68, 2020-11-03:
Arduino: 1.8.13 (Windows 10), Board: "Arduino Uno"

grblUpload:27:10: fatal error: grbl.h: No such file or directory

 #include <grbl.h>

          ^~~~~~~~

compilation terminated.

exit status 1

grbl.h: No such file or directory

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

tanpa-aran, 2020-11-13:
I have same issue

MeJasonT, 2020-11-15:
looks like the arduino ide cant find the files associated with the grbl master ino.
in your documents folder you may have a folder called arduino in there will be a folder called libraries the grbl folder should be in there. it should look somthing like 
[image]

from the arduino ide
select examples, find grbl and open the grbl project

[image]

if the associatd files are missing you get the error you are observing
Im not saying this is the issue but this may be helpful if you dont have this set up properly

quotelawrence, 2020-12-22:
sorry to ask this here if it is the wrong location though I have had a major catastrophe with my 3018 laser engraver I tried to update firmware through lasergrbl here is the screen shot, is there anything I can upload /download to help reestablishing communication 

[image]

semlanik, 2020-12-26:
@quotelawrence make sure that you selected correct bootloader type for your arduino

quotelawrence, 2020-12-31:
> @quotelawrence make sure that you selected correct bootloader type for your arduino

Thank you I use xloader and it worked now new issue no power going to laser x & y work perfectly

alimorshedy, 2024-10-23:
> > @quotelawrence make sure that you selected correct bootloader type for your arduino
> 
> Thank you I use xloader and it worked now new issue no power going to laser x & y work perfectly

i need .hex file