gnea/grbl #955: After setting steps/mm settings, work coordinates can't be set, movement is unpredictable

breiler, 2020-11-09:
After changing any of the step settings ($100, $101, $102) changing work positions can't be issued. Any movement command issued after will behave unpredictable the first time.

Steps to reproduce:
* Move the machine to a non zero machine coordinate
* Change the $10X setting from 250 to 100: `$100=100`
* The work coordinate is updated to reflect the new step configuration
* Issue command to move relatively 0mm: `G91 G0 X0`
* The machine moves to absolute position X0 but should

[image]

I expect that work offsets should be possible after changing settings. I would also expect the machine to honor the relative movement command to the current position.

$I
```
[VER:1.1h.20190825:]
[OPT:V,15,128]
```

$$
```
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 100.000    (X-axis travel resolution, step/mm)
$101 = 250.000    (Y-axis travel resolution, step/mm)
$102 = 250.000    (Z-axis travel resolution, step/mm)
$110 = 10000.000    (X-axis maximum rate, mm/min)
$111 = 10000.000    (Y-axis maximum rate, mm/min)
$112 = 10000.000    (Z-axis maximum rate, mm/min)
$120 = 200.000    (X-axis acceleration, mm/sec^2)
$121 = 200.000    (Y-axis acceleration, mm/sec^2)
$122 = 200.000    (Z-axis acceleration, mm/sec^2)
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
ok
```

terjeio, 2020-11-10:
The current step count (integer) and the parser/planner positions (float) are not synced on a change of steps/mm. IMO after changing such a basic setting the controller should always be hard reset.

A fix could be to update the step count to match the parser position on changes?

breiler, 2020-11-11:
Thanks @terjeio, I'll update the wiki that a soft reset is required after some settings are made to prevent any unintentional movement causing damage to the machine. In UGS we will try to do a reset when some of these settings have been altered.