gnea/grbl #956: Laser Mode Enable (Pin 11) not responding on either arduino

daviddanielparker, 2020-11-10:
Hey guys, I'm new to this, so bear with me:

I'm a student at a carpentry training center where my instructor plopped a JTech 2.5W Laser setup in front of me, running into a CNC shield with an Arduino Uno. The last student to try to "get it working" burnt out our laser diode and fried one of the CNC shields by wiring it wrong. I put it back together with a new CNC/Uno combo, but my first few test runs didn't turn out so hot (included image, because it made a sweet phone wallpaper). 

[image]

The laser started fading and eventually sputtered out to the point where it wouldn't fire at all. The customer support at JTech said it was most likely the laser diode itself, and he sent me a new one, which I have installed now.

After a few days of quick-learning laser software, I was able to run a few programs (the laser works fine now), except the PWM is non existent. CW clicks on (manual mode), but TTL doesn't respond to M3 or M5. The Gcode I tried to run, suggested by JTech, is G1 F100 M3 S1000. Nothing happens. The software I've found to work best for my uses is LaserGRBL. I can jog the motors on the machine just fine, and I can run a program and turn on the manual mode, but it's unacceptable for what the lasers' capabilities. 

JTech told me the Uno board should read a voltage of 4.7-5 V at pin 11, which apparently controls the PWM. Instead of 4.7, I get nothing. 0. Not even a blip. Does anyone have any idea what could be causing this short?
Note: we have 2 arduino boards, one was brand new when I replaced the old one, and both boards read 0V at pin 11. 

[laser burn error pin 11.zip](https://github.com/gnea/grbl/files/5517871/laser.burn.error.pin.11.zip)

Help?

doppelhub, 2020-11-30:
Just to make sure, I assume you're using grbl 1.1+?  I ask because pin11 used to do something else (limit switch).  Disconnect the shield, connect a voltage meter to ground and pin11 (on the uno), then connect to grbl and type the following lines:
$30=1000
$X
$M3
S1000

Note that "$30=1000" will change whatever value was previously set.  This sets the maximum integer value for the PWM output on pin 11.  When we later send "S1000" that means "output maximum PWM value (5V 100% of the time) on pin 11."

daviddanielparker, 2020-12-01:
Hey I did figure it out! My problem ended up being a bad wire between my
driver and the GRBL. Thank you!