gnea/grbl #957: Laser power

rbandeira-pro, 2020-11-16:
Hi,
I'm very new in the laser engraving. I've just assembled my new 10W Laser CNC and started testing it.
After reading lot of information I'm getting in the GRBL parameters and settings. (laserGRBL)
My problem is that my laser doesn't work as it's expected to be.
I'm not sure if I'll ask the good questions but I need help.
Here is the model I bought: 
https://www.amazon.fr/gp/product/B07Z4Z4RDV/ref=ppx_yo_dt_b_asin_title_o07_s01?ie=UTF8&psc=1
in a 10000mW.
Here below where I am from now:
* The laser light turns ON but it doesn't engrave as it is supposed to engrave. It seems to be powerless. 
I tested in a very light wood and cardboard.
I've tested many settings:
*** the distance from the lens to the surface: I tested with 2 cm and with 5 cm trying to figure witch focal I have to use. I can turn the lens and I quite sure I tuned the good focusing with a very tiny dot.
*** I changed the laser mode ($32 in 1 and 0) using different images for a grey levelling or just a ON/OFF setting

The result is always the same: sometimes it leaves just a small mark from the beginning then it fades after a moment.

** Now I went in a strange situation where it worked for a non failure engrave, but the settings are very strange:
speed=30mm/s
M4 mode and M5 to stop
minimum=0 and maximum power=500 (instead of 1000)
And the distance for the focal 2 cm.
With the 500 (50%) off the power it seams to engrave in a good strength, I'm not sure it is the power at 100% but it worked.
If I change my maximum value it doesn't engrave any more. Neither if I set more or less then 500.

Can somebody help me to find what am I doing wrong, please?

Here is my GRBL configuration
************************************
$0=10 (Step pulse time)
$1=25 (Step idle delay)
$2=0 (Step pulse invert)
$3=2 (Step direction invert)
$4=0 (Invert step enable pin)
$5=0 (Invert limit pins)
$6=0 (Invert probe pin)
$10=2 (Status report options)
$11=0.010 (Junction deviation)
$12=0.002 (Arc tolerance)
$13=0 (Report in inches)
$20=0 (Soft limits enable)
$21=0 (Hard limits enable)
$22=0 (Homing cycle enable)
$23=0 (Homing direction invert)
$24=25.000 (Homing locate feed rate)
$25=500.000 (Homing search seek rate)
$26=250 (Homing switch debounce delay)
$27=1.000 (Homing switch pull-off distance)
$30=1000 (Maximum spindle speed)
$31=0 (Minimum spindle speed)
$32=1 (Laser-mode enable - THIS Setting has been tested)
$100=80.000 (X-axis travel resolution)
$101=80.000 (Y-axis travel resolution)
$102=800.000 (Z-axis travel resolution)
$110=10000.000 (X-axis maximum rate)
$111=10000.000 (Y-axis maximum rate)
$112=10000.000 (Z-axis maximum rate)
$120=500.000 (X-axis acceleration)
$121=500.000 (Y-axis acceleration)
$122=500.000 (Z-axis acceleration)
$130=200.000 (X-axis maximum travel)
$131=200.000 (Y-axis maximum travel)
$132=200.000 (Z-axis maximum travel)
**************************************************

Best regards.
RBandeira

Mungbeanz, 2021-01-08:
$30=1000 (Maximum spindle speed)
this should refer to your PWM setting.  So 500=50% and 1000=100%.  Match your gcode software to this.
I personally use 255 as my max PWM setting.  This shouldn't matter but may be worth a try.  Remember to update your gcode software to reflect this

quotelawrence, 2021-01-29:
I am also going through a similar issue I have wired the laser module to the board and as soon as I plug the power supply in the laser comes on full strength I though there was possibly a setting out of whack as this machine was running fabulous and then I shorted out the board replaced it could not get anything working so I purchased another laser module and as I said I am here now, but I did find something quite interesting I have a special button and I use it for focusing, so I put the $30=5, and I put 31=0,  in the button I put m3=1 m5=0, and as I was manipulating the buttons I could see the beam actually change in intensity so I believe everything is working I just need to know are some laser module or boards capable of working in minus state of power or a different code to get it to shut off

quotelawrence, 2021-01-29:
what happened to my comment

Mungbeanz, 2021-01-30:
> I am also going through a similar issue I have wired the laser module to the board and as soon as I plug the power supply in the laser comes on full strength I though there was possibly a setting [...]

Not too sure what you are wanting to achieve with such low gcode PWM settings. 0-5 is very very coarse. You sill have no greyscale ability at those settings.  If your driver works with "reverse" logic you can always recompile grbl editing config.h
Un-commenting the last line and USE_SPINDLE_DIR_AS_ENABLE_PIN

> // Inverts the spindle enable pin from low-disabled/high-enabled to low-enabled/high-disabled. Useful
// for some pre-built electronic boards.
// NOTE: If VARIABLE_SPINDLE is enabled(default), this option has no effect as the PWM output and
// spindle enable are combined to one pin. If you need both this option and spindle speed PWM,
// uncomment the config option USE_SPINDLE_DIR_AS_ENABLE_PIN below.
** #define INVERT_SPINDLE_ENABLE_PIN // Default disabled. Uncomment to enable.**

Hardware can also be used:
https://github.com/gnea/grbl/issues/22#issuecomment-341931756

treesurf, 2022-11-30:
grbl v1.1h in M4 dynamic laser power mode seems to base it's power mode on motor pulses not mm distance.
On my router Y would cut through but X only cut about half depth.
Pulses/mm on X motor was about 2X pulses/mm on Y motor because of gearing,
with the same pulses/revolution on each motor driver. 
I doubled the pulses/revolution on Y motor and doubled $101= setting in grbl
so X and Y pulses/mm are now about the same.
I now get X and Y both cutting through.