gnea/grbl #958: is it possible to have homing / call homing cycle if my axis are not set up with the 'hand cartesian rule'

KevinBaileyCrum, 2020-11-16:
My system has a peculiar drive such that the y and z axis are sharing the same belt.  Therefore if I want to move 
`y+1` this equates to be `y+1, z+1` If I want to move `z+1` then I move `z+1` and `y+1` etc.

My question is, is it possible to still use homing?  
Is there a grbl command to represent home that I can call in a stream?

for example I can manually stream the gcode to max out the x,y,z axis in the way I know my machine has to move. If I wrote a function that sent gcode to move (in my weird way) is there a way to hit a limit switch and tell grbl my machine is at home?

KevinBaileyCrum, 2020-11-16:
I understand this is less an issue and more of a question but I cant find a grbl forum

langwadt, 2020-11-16:
isn't just a COREXY setup? except that it's usually x/y combined

KevinBaileyCrum, 2020-11-17:
good to know about COREXY. never heard of them.  Yes my system is like that but 3d.  Basically I'm asking if there is a way to still use homing/limit switches.  

so i have code that is sending grbl GCODE through serial communication, much like simple_stream.py but im sending gcode rather than a file.  I want to be able to write my own HOME function that relates to my axis and then send the homing done alert to grbl so that it is not blocked.

terjeio, 2020-11-17:
In [my port](https://github.com/terjeio) of grbl to 32-bit I have ripped out the coreXY code from several files and moved it to a [separate file](https://github.com/terjeio/grblHAL/blob/master/grbl/corexy.c). To make it easier to add new kinematics I have also defined an [internal API](https://github.com/terjeio/grblHAL/blob/master/grbl/kinematics.h) that allows that without changing existing code (except for a single call for setup) . Code for [wall plotter](https://github.com/terjeio/grblHAL/blob/master/grbl/wall_plotter.c) and [Maslow](https://github.com/terjeio/grblHAL/blob/master/grbl/maslow.c) kinematics has been added using this API but is as of now untested.

Adding code for this API could be an option for you?

KevinBaileyCrum, 2020-11-17:
I think i devised a solution.  To address my last paragraph I was looking for https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#feedback-messages specifically `[msg:'$H'|'$X' to unlock]` 

therefore I can send my gcode mapped to fit my odd drive system and if I hit a limit switch or have homing enabled I can simply send $X or $H respectively.

KevinBaileyCrum, 2020-11-17:
oh darn that wont work because I cant know what axis hit the limit swithc all I am getting is alarm:1 but not a message on what axis hit

terjeio, 2020-11-18:
>  I am unable to send '?' in the locked state.

Why not? If able to send but no reply is forthcoming you can fix grbl to do what is intended(?) with a few lines of code. 

https://github.com/gnea/grbl/blob/eefe2bb95bb7b21ec2bb87e6ab6e20747e1626c4/grbl/protocol.c#L226-L238

Add these lines in the do...while loop...

https://github.com/gnea/grbl/blob/eefe2bb95bb7b21ec2bb87e6ab6e20747e1626c4/grbl/protocol.c#L249-L253

...to do what the comment "// Block everything, except reset _and status reports_" states and you'll get the report?

Or is there not enough flash available for that?

From my 32-bit port:
```
        if ((alarm_code_t)rt_exec == Alarm_HardLimit || (alarm_code_t)rt_exec == Alarm_SoftLimit || (alarm_code_t)rt_exec == Alarm_EStop) {
            system_set_exec_alarm(rt_exec);
            grbl.report.feedback_message((alarm_code_t)rt_exec == Alarm_EStop ? Message_EStop : Message_CriticalEvent);
            system_clear_exec_state_flag(EXEC_RESET); // Disable any existing reset
            while (bit_isfalse(sys_rt_exec_state, EXEC_RESET)) {
                // Block everything, except reset and status reports, until user issues reset or power
                // cycles. Hard limits typically occur while unattended or not paying attention. Gives
                // the user and a GUI time to do what is needed before resetting, like killing the
                // incoming stream. The same could be said about soft limits. While the position is not
                // lost, continued streaming could cause a serious crash if by chance it gets executed.
                if(bit_istrue(sys_rt_exec_state, EXEC_STATUS_REPORT)) {
                    system_clear_exec_state_flag(EXEC_STATUS_REPORT);
                    report_realtime_status();
                }

                grbl.on_execute_realtime(STATE_ESTOP);
            }
            system_clear_exec_alarm(); // Clear alarm
        }
    }
```

KevinBaileyCrum, 2020-11-20:
So idk if this is in the docs or not but I figured out how to connect after the alarm.
- Essentially with `$21=1` and when my machine hit a limit ('to simulate homing ie Move one axis at a time until a switch is hit and assume the switch that was hit was the axis I was moving -- not uber uber safe but it works for now)
- grbl sends `ALARM: 1` and then `[MSG:Reset to continue]`  
- it is this reset to continue state that blocks everything even sending my `$X` command.  It is not until I send `'\x18\n'` for the reset command (this is the part that I couldnt find in the wiki) AND THEN I am able to send `'$X\n'` to unlock.

in my reading of the wiki it seemed like $21=1 leads to if limit hit -> alarm -> send $X to unlock.  Not the case in my situation whereby its alarm -> ^X -> $X -> continue

Hope this helps someone else.  If this is present in the wiki let me know I'd love to reference it, otherwise I may try type an edit somewhere for the wiki to disambiguate that.   Leaving issue open for now to see if the ^X is in the wiki, I understand the title doesn't fully match the issue Im addressing ('cant send $X to unlock after hard limit hit') might be more appropriate.  Hope this helps.

closed 2020-11-25 by KevinBaileyCrum