gnea/grbl #961: Hard Limit Triggered, Motors loose holding torque and drop Z axis.  Way to avoid?

KevinBaileyCrum, 2020-11-25:
In grbl when I have hard limits enabled and it a limit switch the machine enters alarm state.
In the time it takes to send \x18 and then send $X the holding torque is dropped from the motors.  
As a result my z-axis is dropped!

Is there a setting in grbl that I can tinker to maintain holding torque during alarm and reset?  I tried looking over config.h, limits.c, and motion_control.c but am having trouble finding what I am looking for

-note: I understand this is less of an issue and more of a question but I cannot find the existence of a grbl question forum so I am posting here.  If this is incorrect, my apologies.

terjeio, 2020-11-25:
$1=255?

KevinBaileyCrum, 2020-11-25:
$1 is at 255

doppelhub, 2020-11-30:
@terjeio: Setting $1=255 isn't going to keep the steppers enabled during an alarm condition.
@KevinBaileyCrum: Yes, you can change this behavior in firmware*.  If you're not familiar with grbl, A simpler method might be to disconnect the stepper enable pin (pin8 on the 328p Uno)... that'll just always leave the steppers enabled.

*Search grbl source for "st_wake_up()" for the implementation.

KevinBaileyCrum, 2020-12-01:
@doppelhub I'll take a look at `st_wake_up()`  I am unsure when reading the source code because i certainly want to stop the steppers from moving but I dont want to disable their holding torque.  I was thinking it would involve changing `mc_reset()` or `st_go_idle` but again namely want to stop the motors for moving.

in regards to disconnecting the pin from the arduino, would limit switches still work / send signals to halt motors in grbl but just not kill the signal on the duino?

KevinBaileyCrum, 2020-12-02:
ended up changing the st_go_idle() method.  it works for my implementation whereby the limit switch is hit and the motors do not drop their holding torque and therefore keeping my z axis suspended in the air.  Hope this isnt too dangerous of a change and helps some else out. 

```
// Stepper shutdown
void st_go_idle()
{
  // Disable Stepper Driver Interrupt. Allow Stepper Port Reset Interrupt to finish, if active.
  TIMSK1 &= ~(1<<OCIE1A); // Disable Timer1 interrupt
  TCCR1B = (TCCR1B & ~((1<<CS12) | (1<<CS11))) | (1<<CS10); // Reset clock to no prescaling.
  busy = false;

  // Set stepper driver idle state, disabled or enabled, depending on settings and circumstances.
  /* bool pin_state = false; // Keep enabled. */
  /* if (((settings.stepper_idle_lock_time != 0xff) || sys_rt_exec_alarm || sys.state == STATE_SLEEP) && sys.state != STATE_HOMING) { */
    // Force stepper dwell to lock axes for a defined amount of time to ensure the axes come to a complete
    // stop and not drift from residual inertial forces at the end of the last movement.
    /* delay_ms(settings.stepper_idle_lock_time); */
    /* pin_state = true; // Override. Disable steppers. */
  /* } */
  /* if (bit_istrue(settings.flags,BITFLAG_INVERT_ST_ENABLE)) { pin_state = !pin_state; } // Apply pin invert. */
  /* if (pin_state) { STEPPERS_DISABLE_PORT |= (1<<STEPPERS_DISABLE_BIT); } */
  /* else { STEPPERS_DISABLE_PORT &= ~(1<<STEPPERS_DISABLE_BIT); } */
}
```

closed 2020-12-02 by KevinBaileyCrum