gnea/grbl #963: Probing routine

arlimbad222, 2020-11-28:
I made a 6mm touch probe for my machine & now I want zero my work offset at centre of circle please give me advice for probing routine I am very new to this .

terjeio, 2020-12-04:
A search either here or generally on the internet will find the advice you need?

Some grbl senders provide advanced probing functionality such as finding the center of a hole. Issue #931 has links and links to links for you to explore, you may even find usable source code if you want to implement the routine yourself.

arlimbad222, 2020-12-04:
Hey terjeio thank you so much for reply 

 i write some probing routine for corner zero i use 6mm probing tool for this but my question is not find corner of the part i want to touch my probe 4 side of hole and automatic find center of the circle .

Here is my code to find front right corner of the square part . i just need to set my tool approx corner of my part and boom this code will find very accurate corner . so my question is can i write this type of code for find center of the hole or any circular part . 

G91 G38.2 Z-20 F30

G90 G10 L20 P0 Z0

G91 G0 Z2

G91 G0 X-5 ←

G91 G0 Y-10 ↑

G91 Z-5

G91 G38.2 Y15 F30 ↓

G90 G10 L20 P0 Y-3

G91 G0 Y-3↑

G91 G0 X15 →

G91 G0 Y15 ↓

G38.2 X-15 F30 ←

G90 G10 L20 P0 X3

G91 G0 X2 →

G28 G91 Z0
G90
G28 G91 X0 Y0
G90

terjeio, 2020-12-04:
You need to read back the probed positions and calculate where the center is, AFAIK its not possible to achieve this by gcode only.

In my sender I have this user interface for center probing:

[image]

It shows a preview of the gcode I use for probing a 20mm hole, the `!` in front of some commands is used internally to tell the algoritm I use to drop the previous probing result since I probe an edge twice with different speeds.

Code is [here](https://github.com/terjeio/Grbl-GCode-Sender/blob/master/CNC%20Controls%20Probing/CNC%20Controls%20Probing/CenterFinderControl.xaml.cs), after collecting the four positions required then the center is calculated in `OnCompleted()`. Perhaps this is not much help for you but is what I have at hand.

closed 2021-01-11 by arlimbad222