gnea/grbl #964: Error compiling only when #define ENABLE_DUAL_AXIS is enabled and Y Axis selected

trentaw, 2020-11-29:
C:\Users\trent\Documents\Arduino\libraries\grbl\report.c: In function 'report_realtime_status':

C:\Users\trent\Documents\Arduino\libraries\grbl\report.c:654:1: error: expected declaration or statement at end of input

 }

I only get the error with #define DUAL_AXIS_SELECT set to Y_AXIS

mikerace3030, 2021-01-15:
I have this same issue. I have tried on a Mega and a UNO. 

Arduino: 1.8.13 (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\racema\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -vid-pid=2341_0042 -ide-version=10813 -build-path C:\Users\racema\AppData\Local\Temp\arduino_build_771830 -warnings=none -build-cache C:\Users\racema\AppData\Local\Temp\arduino_cache_371261 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\racema\Documents\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\racema\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -vid-pid=2341_0042 -ide-version=10813 -build-path C:\Users\racema\AppData\Local\Temp\arduino_build_771830 -warnings=none -build-cache C:\Users\racema\AppData\Local\Temp\arduino_cache_371261 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\racema\Documents\Arduino\libraries\grbl\examples\grblUpload\grblUpload.ino

Using board 'mega' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr

Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr

Detecting libraries used...

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "C:\\Users\\racema\\AppData\\Local\\Temp\\arduino_build_771830\\sketch\\grblUpload.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for grbl.h: [grbl]

ResolveLibrary(grbl.h)

  -> candidates: [grbl]

"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\mega" "-IC:\\Users\\racema\\Documents\\Arduino\\libraries\\grbl" "C:\\Users\\racema\\AppData\\Local\\Temp\\arduino_build_771830\\sketch\\grblUpload.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Using cached library dependencies for file: C:\Users\racema\Documents\Arduino\libraries\grbl\coolant_control.c

Using cached library dependencies for file: C:\Users\racema\Documents\Arduino\libraries\grbl\eeprom.c

Using cached library dependencies for file: C:\Users\racema\Documents\Arduino\libraries\grbl\gcode.c

Using cached library dependencies for file: C:\Users\racema\Documents\Arduino\libraries\grbl\jog.c

Using cached library dependencies for file: C:\Users\racema\Documents\Arduino\libraries\grbl\limits.c

Using cached library dependencies for file: C:\Users\racema\Documents\Arduino\libraries\grbl\main.c

Using cached library dependencies for file: C:\Users\racema\Documents\Arduino\libraries\grbl\motion_control.c

Using cached library dependencies for file: C:\Users\racema\Documents\Arduino\libraries\grbl\nuts_bolts.c

Using cached library dependencies for file: C:\Users\racema\Documents\Arduino\libraries\grbl\planner.c

Using cached library dependencies for file: C:\Users\racema\Documents\Arduino\libraries\grbl\print.c

Using cached library dependencies for file: C:\Users\racema\Documents\Arduino\libraries\grbl\probe.c

[... cut ...]
Using previously compiled file: C:\Users\racema\AppData\Local\Temp\arduino_build_771830\libraries\grbl\serial.c.o

C:\Users\racema\Documents\Arduino\libraries\grbl\report.c: In function 'report_realtime_status':

C:\Users\racema\Documents\Arduino\libraries\grbl\report.c:654:1: error: expected declaration or statement at end of input

 }

 ^

Using library grbl in folder: C:\Users\racema\Documents\Arduino\libraries\grbl (legacy)

exit status 1

Error compiling for board Arduino Mega or Mega 2560.

ZASto, 2021-01-20:
Funny that no one answered.
There is a closing curly bracket missing in report.c file. Add one as pr this screen shot.
[image]

mikerace3030, 2021-01-20:
Thanks!, I will give that a try tonight.

aapicella, 2021-03-19:
I looked at that code.  Seems like its like 600 that need the }  the If in the   if in line 571 isn't terminated.

JipeB, 2024-01-06:
Hi, (January 2024...)
The file is still not fixed and triggers a compilation error when dual axis is validated.

The error is in the report.c file line 581; The closing brace is missing.

         #if (DUAL_AXIS_SELECT == Y_AXIS)
            if (bit_istrue(lim_pin_state,bit(X_AXIS))) { serial_write('X');  **}** <<<<<<<<<<
            if (bit_istrue(lim_pin_state,(bit(Y_AXIS)|bit(N_AXIS)))) { serial_write('Y'); }
          #endif

Add the missing brace and fix the end of the file which should be like below.

  #endif

  serial_write('>');
  report_util_line_feed();

  #ifdef DEBUG
    void report_realtime_debug()
    {

    }
  #endif
}

JP