gnea/grbl #968: TB6600 drivers not working with grbl

jcalcote, 2020-12-13:
What's the secret to getting grbl to signal TB6600 stepper drivers properly? I can drive them just fine using my own arduino program, but grbl seems not to know how. I've played with every setting I can think of (but I have not modified the config.h settings). Nothing works. When I run a G command the motors lock and hum, but no motion occurs.

jcalcote, 2020-12-14:
Update: I inched my way little by little to a point where I'm able to jog and G0 the steppers to various positions at various feed rates. What I had to do was set the stepper drivers to 16x microstepping, and crank the current to the steppers' max. I have one 2.8 nema 23 and one 3.0 nema 23 (the X axis is a little heavier, having to pull the entire Y axis carriage). This allowed me to set the max feed rate at 1600 mm/s for both axes. I can now run G0 X/Y F1600 commands and it moves just fine. With the proper feed, I can get it to jog correctly also. 

What's still broken is homing. When I use $H (or use the home button in UGS), the motors just hum until grbl gives up and sends an error back that it must be well outside the range now. 

I think it's weird that I have to set the drivers to micro-step mode - full-step mode won't work at all unless I've got the speed cranked way down. I don't understand why it doesn't home. Please don't ask me to check the limit switches - I already know those work fine because while it was moving with a G0 command, I hit the switches manually and grbl responded with an error. Also grbl is not alarming about limits.

It's taken me literally days to get this far. It's ridiculous that no one out there has tried this and had anything like the same issues I've had.

One final thought - the title of this issue is TB6600, but the fact is, I have the "new improved" version of the common clone driver built around this chip. However, the new improved version is not actually based on the TB6600, but rather on the TB67S109AFTG - here's the one I bought:

https://www.amazon.com/gp/product/B088CRCMP7/ref=ppx_yo_dt_b_asin_title_o07_s01?ie=UTF8&psc=1

If anyone has any ideas - please respond.

jcalcote, 2020-12-14:
Update: I found the problem, and I'm embarrassed to say it was at least partially related to limit switches - well - to those on a particular axis. You see, I neglected to provide a critical bit of information: I'm building a laser engraver, so my system only has two axes. The problems were as follows:

1. I needed to update config.h to handle homing on a two-axis system. When attempting to home, the system was configured to home the z-axis first to clear the work space, then to home the x and y axes. Since I had hard limits enabled, grbl could never get the z axis to trigger it's homing limit, so it just hung there.

2. I had wired the limit switches to normally closed, but neglected to take into account that the z axis was open. Therefore when grbl started up, it always had a limit fault. 

If it's ok with the community, I'd like to add a page to the wiki called "Two-Axis System Considerations" and write all this up. Then I'd like to reference this new page from the Laser Grbl page.

closed 2020-12-14 by jcalcote

jcalcote, 2020-12-14:
PS: Wiki changes completed.

makarandkarbhari, 2022-09-12:
Hi @jcalcote,
I have been working on a similar project of running 2 stepper motors for 2 axes (x & y) using Arduino mega and 2 tb6600 drivers for each motor, I am not sure what is wrong either it is my grbl settings or else something is wrong when I set them up in IDE. However, I am running it on a full-step mode(which could be a reason from the above discussion). I also don't have any limit switches for now but I am planning to buy them in the near future. 

Please guide me with some light on this topic or is it possible for you to send that config.h file to me

Thank you

SmokinPete55, 2025-01-24:
Hi @jcalcote 

I have made my own Laser engraver and previously used an Arduino with a CNC Shield on it. After working for several months, I began to burn up power supplies (2) and went through two controller cards. I had recently purchased a different controller card but tech support says it won't work with a Mac at all and I would also need to operate the card with a windows 7 machine. Absolutely ridiculous as there are probably very few out there and are no longer supported by Microsoft.

I have since gone back with the shield but have integrated TB6600s for X-Axis and Y-Axis. I have a basic schematic drawn up and want to find out if there are driver files for the TB6600s? Query on internet says yes, but I have yet to find any. Before I tear everything apart and reset the Arduino/CNC Shield in my case. 

Any assistance would be greatly appreciated. Thanks in advance!!!

[image]

All but last wiring of motor connections is neatly routed. This is where I ran into problems and the manufacturing company gave me the info above.