gnea/grbl #976: Please clarify proper use of HOMING_SINGLE_AXIS_COMMANDS - does not work?

arminlinder, 2020-12-30:
Hi,

After enabling the setting config.h, compiling and uploading the commands $HX and $HY still do not work, I get "error:3” : “Grbl ‘$’ system command was not recognized or supported.”, just like before when HOMING_SINGLE_AXIS_COMMANDS had not been enabled. BUT, if I get the machine state (?), it shows  [homing ....], but no axis actually moves.

Details:

 I have a two axis setup, with an NC switch at each axis home position. Beeing able to home each axis individually would help me adjusting and testing limit switches. Everything seems setup properly, I can run both axis using a terminal application, and $H does correctly home both axis as defined in the homing cyle, which reads HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS)). Trying $HX or $HY gives "error:3” : “Grbl ‘$’ system command was not recognized or supported.”, which makes sense.

With HOMING_SINGLE_AXIS_COMMANDS enabled, still neither $HX nor $HY does work. If I try them, I still get "error:3”, but, unlike before, "?" shows a [homing ...] feedback message along with some corrdinates, which, however, don't ever change, and no axis actually moves. Huh?

$H still works correctly like before.

Any ideas welcome.

Thnx, Armin.

arminlinder, 2020-12-31:
Stop! Before anyone wastes time on this -- I tried the setting on a 3 axis machine today, and they worked like a charm.

Theory #1: is there a problem with the command on 2 axis machines?
Theory #2: despite I checked everything twice, I oversought an error on my 2 axis machine?

I'll investigate, and comment as sonn as I am sure it's indeed a GRBL problem.

Armin.

ablankenship10, 2021-01-06:
I'm having the same issue on 1.1h. I made the change in the correct place (in Arduino library copy of the config.h file) and when I ran Verify in Arduino IDE I noticed that the compiled size was slightly larger with the HOMING_SINGLE_AXIS_COMMANDS variable defined, telling me that it should in fact be recognized as changed.

Using LaserGRBL i still receive the Setting Disabled error when attempting to run either $HX, $HY, or $HZ.

arminlinder, 2021-01-07:
For completeness:

- I have 1 switch per axis only

As promised, I tried to reproduce my original problem today, and could not. I reset the machine to 2 axis, and left HOMING_SINGLE_AXIS_COMMANDS uncommented, but the original problem (error: 3 if $HX or $HY) did not show up again. Instead, both commands work like intended. $H homes the axes like defined in config.h, $HX and $HY (and $HZ) home individual axes.

Closing this issue, it seems that I made some mistake somewhere, but I cannot reproduce it to find out. 

Everything's fine, closing this issue as PEBCAC.

Armin.

closed 2021-01-07 by arminlinder