gnea/grbl #977: Spindle pause

MARIOBASZ, 2020-12-31:
It is practical for me to add an immediate action command pause spindle together with pause, for example to clean a bit of material stuck in the middle of multiple holes

MeJasonT, 2020-12-31:
use the door overide, it pauses the job, retracts the tool and stops the spindle. you can resume by clearing the alarm and pressing cycle strat - the program continues from where it left off.  You dont need to have a safety door, you can use a switch on the program pause connection (as you would wire in an external pause button) but enable the door function which will add a full retract and stop.  Pro machines leave the tool engaged in the work but reduce the spindle speed down to a slow rpm say 200.  I actually prefer the grbl method

MeJasonT, 2020-12-31:
Happy New Year

MARIOBASZ, 2020-12-31:
Thanks!!!!  I did not know it

closed 2021-01-01 by MARIOBASZ

harryberlin, 2021-01-01:
Does grbl understand the M0 command?

MARIOBASZ, 2021-01-01:
the goal is to interrupt briefly in the middle of work and then continue. I'm going to take the advice and try open-door interruption.

MeJasonT, 2021-01-01:
using a progrmmed code option would mean that you intended to do a clean on a regular basis and not a random choice as and when required