gnea/grbl #979: GRBL config for Stepper based Galvo

TehseenHasan, 2021-01-02:
I am using GRBL for my DIY diode-based Laser Engraver and It is working great. Now I was planning to make a **Galvo based laser engraver** using the same 2.5W 450nm diode laser. And for a low-cost option, I found that I can actually make the **Galvos using Nema 17 stepper motors**. For example, like this guy did : [Link](https://www.thingiverse.com/thing:3083774)
Now my question is that how can I configure the GRBL to work for this stepper based Galvo configuration the same way as the normal XY Plotter configuration works. e.g. what should I set the X and Y travel limits (because the movement of steppers will very narrow in galvos) and other parameters of GRBL?
Thanks!

[image]

[image]

https://youtu.be/bdVrkIBskMs

TehseenHasan, 2024-11-06:
> bu konuda bir ilerleme kaydedebildin mi?

Unfortunately, I couldn’t continue with the project due to other commitments, and it has now been closed. As far as I know, no one has developed this idea online yet.

Pathogenex, 2025-04-21:
I know you've given up but I also had the same idea and ended up here.

Pretty simple setup once you get it. 

2 motors X and Y need to be the XY in grbl (X horizontal, Y vertical) 
So the mirror that goes vertical is Y and the mirror that goes left/right is X

There a few considerations you haven't thought of yet. 
And that is an "F-Theta lens"... 
This is the biggest crux for a DIY Galvo CO2 laser as the lenses cost about 150. They are needed to maintain beam focus along a flat plane and you need to get the doped version for the light wavelength your laser is using. Otherwise you're going to make an expensive trinket... 

You could easily run a few tests with a low power red laser pointer. Home your machine and set each mirror position to exactly a 45 angle (IE so the laser is pointed straight ahead)

Then focus the laser pointer on that position. 
Then measure the dot radius after moving each mirror an identical number of steps. 
Keep doing this measuring until the radius is at an unacceptable focus. You then have you machine cutting area. 

Likely the beam deviation will be rather high.. This is where a theta lens comes in useful as you would place this after the galvo mirrors and it would sustain the focus point along a flat plane. 

CO2 infrared lenses are generally expensive and I can't say I have ever seen much variation in the types available. You may be able to build your own Theta lens but the knowledge on how to do that seems to a rather guarded secret. 

Hopefully this helps someone in the future.