gnea/grbl #980: Z Axis corrupts after about 24h work and breaks the bit

benclewett, 2021-01-03:
Forgive if this is not the correct place.  

On a big carve, my 3018 1.1f board works fine for about 24h, or about 15,000 lines into the GCode file.  Then the Z-axis will corrupt and dive the bit into the work, often breaking the bit.  It's not always the same place, but around this time.  See the attached image.

I've tried various software including Candle, Easel, and the supplied offline controller, which all do the same.

I've tried several different cuttings (GCode files), the behaviour is consistent.  Always down by about 6mm after 24h.

I've checked the GCode (all 22,000 lines or so), and the z positions fed by the controller are nowhere in the code, as confirmed by Candle.

Why could my controller be doing this?  How can I stop it?  Is this a problem with this code, or with something else?

Thanks,

Ben

[image]

ASHIT-AXAR, 2021-01-08:
if you are sure there is no error in the code, then try attaching a fan to z-axis stepper to cool it down during running a job. (also having a fan for stepper drivers is a good idea)

benclewett, 2021-01-08:
Thanks for getting back to me with an idea.

Yesterday my 'Woodpecker CNC CamXTool V3.4" running GRBL, literally went up
in smoke.  I suspect it had cooling issues which may have caused the Z-Axis
controller to fail after a day or two of use.  My supplier (AliExpress)
have kindly offered a replacement. I will see if it's better when it
arrives.  Which I intend to flash with the latest GRBL.

However thanks again for the reply, this was driving me crazy!!

Ben