gnea/grbl #983: Use Servo as Z Axis?

pcace, 2021-01-05:
Hey there, i just found GRBL and tried to find any Info on how i would use a servo as Z Axis?
Is there any way to do that in code?
(i am building a pen plotter and just need up/down for a servo)

Thanks a lot!!!

Elmi77, 2021-01-10:
Can't you use the direction-pin which is on for on direction and off for the other?

svenhb, 2021-02-12:
I like to use: https://github.com/cprezzi/grbl-servo/tree/eggbot/grbl

jhaand, 2022-09-30:
> I like to use: https://github.com/cprezzi/grbl-servo/tree/eggbot/grbl

It would be really nice to merge the servo spindle control with GRBL 1.1 as a build flag. 
I have very serious problems with sending gcode to my pen plotter with GRBL 0.9 and modern software bCNC, gSender, Candle.

svenhb, 2022-09-30:
Do you know this version? https://github.com/cprezzi/grbl-servo

https://grbl-plotter.de/index.php?id=quick-guide&setlang=en

jhaand, 2022-09-30:
@svenhb That looks already better. Unfortunately the grbl-plotter only works under Windows and not Linux. 

But the grbl-servo by @cprezzi already seems to work with GRBL 1.1 
Now to invert the Y-axis with CoreXY.