gnea/grbl #987: Homing

kikirikiki, 2021-01-07:
Hi, I would like to disable the homing seek to have just only one homing cycle, in each axle, either the homing feed or the homing seek. Is it posible to go to the limits just once?

Thanks a lot!

Mungbeanz, 2021-01-08:
config.h
has this option which may be what you are after.
#define N_HOMING_LOCATE_CYCLE 1 // Integer (1-128)

Looks like minimum if 1 cycle post seek.

kikirikiki, 2021-01-11:
and it is not possible to delete this cycle? or delete the seek cycle?

Mungbeanz, 2021-01-12:
Not from a config perspective from what I can see. Could possibly manually edit source code and recompile.  Why do you want to remove the second homing cycle.  It serves to fix problems of inaccuracy during a single homing cycle. ie switch limitations, noise interference etc.