gnea/grbl #988: Spindle pins

Sstaceyh74, 2021-01-09:
I have had a search but I cannot find what I am looking for, sorry if has already been answered.

I have an UNO with GRBL 1.1h the step/dir pins work find and I can move the machine around with UGS, Pin 11 works if send S commands (I can drive it up and down) but other pines are all HIGH... If I send a M3, M4 or M5 the pins do nothing!

Any ideas people

jerseyguy1996, 2021-01-18:
Just a guess for you (as I'm figuring out my own implementation as we speak), but did you remember to uncomment "#define USE_SPINDLE_DIR_AS_ENABLE_PIN"? The section starts on line 352 in config.h.  It reads:

// By default on a 328p(Uno), Grbl combines the variable spindle PWM and the enable into one pin to help
// preserve I/O pins. For certain setups, these may need to be separate pins. This configure option uses
// the spindle direction pin(D13) as a separate spindle enable pin along with spindle speed PWM on pin D11.
// NOTE: This configure option only works with VARIABLE_SPINDLE enabled and a 328p processor (Uno).
// NOTE: Without a direction pin, M4 will not have a pin output to indicate a difference with M3. 
// NOTE: BEWARE! The Arduino bootloader toggles the D13 pin when it powers up. If you flash Grbl with
// a programmer (you can use a spare Arduino as "Arduino as ISP". Search the web on how to wire this.),
// this D13 LED toggling should go away. We haven't tested this though. Please report how it goes!
// #define USE_SPINDLE_DIR_AS_ENABLE_PIN // Default disabled. Uncomment to enable.

Noco2ev, 2025-06-08:
Having the same issue. Have 3 different machines running GRBL on uno's. Blew one up recently and now trying 3 different uno's and not one of them recognizes the M3/M5 commands. I have the #define USE_SPINDLE_DIR_AS_ENABLE_PIN uncommented and as required the VARIABLE_SPINDLE enabled.
Gary