gnea/grbl #989: Clarification needed: limit and homing switches?

Elmi77, 2021-01-10:
Hi,

the description is somewhat unclear for me in terms of the input switches. What I understand

- two limit switches have to be serial wired to work with one input pin
- beside limit switches (which stop motion completely and switch to alarm state, when one is hit) there is also a home switch for initialisation/referencing of the axis

What I do not understand: where do I connect them? There are not enough input pins on an Arduino for both  and the typical pin-out only shows limit-inputs.

So where to connect/define home switches?

Thanks!

PicEngraver, 2021-01-10:
Homing and Hard Limit switches wire to common pins.  Grbl keeps track of what action is appropriately needed.

Mungbeanz, 2021-01-12:
look at the config for hard limit switches.
[https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#21---hard-limits-boolean](url)
This should use your homing switches.  There may need to be some config changes to max limit pins. Depends on how you want to go about things.  I imagine you could just duplicate X and Y for your max limits.  Once the switch is engaged the machine will halt.  Doesn't matter where it is

doppelhub, 2021-01-28:
Search "Connecting Grbl" in the wiki... that'll show you where the limit switches connect.  Note: The limit and homing switches are the same.