gnea/grbl #992: Spiral interpolation in XY axis

sergiv69, 2021-01-12:
Hi,
First of all apologizes if question was already done, I was not able to find it.

It seems like a dummy think but, I can not figure out how to draw spirals (like the snail shell) with X and Y axis. I can only find info on drawing arcs, but it is not an arc, I need spirals.

Is there any command to do this?

I want to do this: https://www.youtube.com/watch?v=7XvDAx_TWLg

SimonJohnCastle, 2021-01-13:
To plot a spiral you simply plot radius x sin() against radius x cos() of the angle and vary the radius by the angle, if you truly want a snail shell then that follows the fibonacci series which is something else.

sergiv69, 2021-01-13:
Hi Simon,

Thank you very much for your answer. But how I can implement that on Gcode? I mean do I have to plot every point I wish with G01 commands, or with G02/03 commands and changing the radius, or there is any better method?

SimonJohnCastle, 2021-01-13:
Unfortunately you will have to plot each point yourself.
May be Inkscape, Gimp or one of the other graphics tools can draw spirals and you can start there.
If you dont know where to start then I could add it as a tool path to my project here [https://www.rcgroups.com/forums/showthread.php?3777429-Conversational-Programming-CAD-Free-CNC](url)

sergiv69, 2021-01-13:
Thanks. I can't open your link, says it was not found, could please re-attach it?

MARIOBASZ, 2021-01-15:
I use bCNC to handle grbl. It has many plugins and from there I generate the figures and I code G. You can generate a plugin to do that (spiral). I think that inside the Trochoidal 3 d plugin there is a spiral; I don't know if somewhere on my computer I have it as a separate plugin. While I am limited with Python, I have been able to make some input.
Options:
Maybe some Cad program has a spiral?
Generate the points and the Gcode from excel?

SimonJohnCastle, 2021-01-15:
Here is a 2 revolution spiral centered on 40,40 with increasing radius 20 by revolution.

In the interest of reducing the points you can draw regular arcs over small distances to get a good result

`
G90 ; ABS

G0 X40.0 Y40.0 ; P1 Fast
G1 Z0 F800
G91 ; REL
G1 Z-1.8 ; DOWN
G2 X0.8 Y1.4 R1.7
G2 X2.1 Y0.2 R3.3
G2 X2.1 Y-1.7 R5.0
G2 X0.8 Y-3.3 R6.7
G2 X-1.6 Y-3.9 R8.3
G2 X-4.2 Y-2.8 R10.0
G2 X-5.8 Y-0.1 R11.7
G2 X-5.7 Y3.4 R13.3
G2 X-3.5 Y6.7 R15.0
G2 X0.6 Y8.3 R16.7
G2 X5.3 Y7.5 R18.3
G2 X9.2 Y4.1 R20.0
G2 X10.8 Y-1.2 R21.7
G2 X9.4 Y-7.1 R23.3
G2 X4.8 Y-11.7 R25.0
G2 X-1.9 Y-13.3 R26.7
G2 X-8.9 Y-11.2 R28.3
G2 X-14.2 Y-5.5 R30.0
G2 X-15.8 Y2.6 R31.7
G2 X-13.0 Y10.8 R33.3
G2 X-6.1 Y16.7 R35.0
G2 X3.2 Y18.3 R36.7
G2 X12.6 Y14.9 R38.3
G2 X19.2 Y6.8 R40.0
G1 Z1.8
G0 Z5 ; UP
G91 ; REL
`

SimonJohnCastle, 2021-01-15:
Here it is executing

[YouTube](https://youtu.be/ItRMob-Ks2c)

sergiv69, 2021-01-18:
Wow, that's impresive. Thank you very much!!!

SimonJohnCastle, 2021-03-20:
I have added the spiral path to my conversational programming tool

[Generating A Spiral Path](https://youtu.be/nijRmX5RPwcl)

There are good reasons why you actually want to do it, as wall as the path being uninterupted

sergiv69, 2021-03-22:
Hi,

Thank you very much! You are very kind.

I was also thinking about the interpolation of a rotary axis and a linear
axis such as the example on this video, see from minute 1:07

https://youtu.be/atVbV22pYN8?t=67

It is incredible what they done! It could be possible to give the order
with standard G01, G02, G03 instructions to do it?

Saludos cordiales / Best regards,
*Sergi Valls*

España

Email
Skype
Web
Youtube

(+34) 932 935 178
(+34) 622 600 690

***@***.***
elite.svals
www.elite.es <http://www.elite.cat/>
www.youtube.com/elitecorte

El medio-ambiente es cosa de todos. Antes de imprimir este Email, piense
bien si es necesario hacerlo.
Este mensaje se dirige exclusivamente a su destinatario y puede contener
información privilegiada o confidencial. Si no es Ud. el destinatario
indicado queda notificado de que la utilización, divulgación y/o copia sin
autorización está prohibida en virtud de la legislación vigente. Si ha
recibido este mensaje por error, le rogamos que nos lo comunique
inmediatamente por esta misma vía y proceda a su destrucción
En cumplimiento de la Ley Orgánica 15/199 de Protección de Datos
Personales, le informamos que sus datos están incorporados a nuestros
ficheros, con la finalidad de mantener nuestra relación comercial. Puede
ejercer los derechos de acceso, rectificación o cancelación de sus datos
mediante escrito dirigido a *Elite Corte Y Afilado, SCP*, en la dirección
de email ***@***.***, acompañando copia de su identificación.

Missatge de SimonJohnCastle ***@***.***> del dia dg., 21 de
març 2021 a les 0:02: