gnea/grbl #993: SAVE G-CODE FILE

arlimbad222, 2021-01-21:
Hey forks 

I am making one horizontal milling & want to run only 25 line of gcode per cycle is there any way to save gcode file on board & run that file with cycle start button push and stop with stop button push. currently i am using one old laptop to send only 25 line of gcode .  and there is no editing riquired just 25 line gcode for life time .

thankyou for help.

SimonJohnCastle, 2021-01-21:
Its a trivial exercise to store a small GCODE in an Arduino Nano and have it sent out on rest or button press.
Or get one of those cheap offline controllers and use that, if your HW supports it already.

Unfortunately you dont say what you are using.

harryberlin, 2021-01-21:
Candle hast custom buttons for cuttom gcodes

arlimbad222, 2021-01-21:
Hey @harryberlin 

What is candle brother I don't know about that think I only use grbl and not an coding guy basically am mechanical engineer can you please explain me breef what is candle ?

SimonJohnCastle, 2021-01-21:
Candle is PC SW for sending GCODE to GRBL, so not what you want.

You can get a Nano clone very inexpensively or use another Uno if you have one, easy to programm, just plug in and go.
If you are new to Arduino programming, its very easy and there are many examples, all you would have to do is send the GCODE over the serial port as strings. To make things easy so you dont have to worry about the responces from GRBL you can put delays between each line to allow the machine to perform its task.

Connect GND, 5V and cross the Rx/Tx lines between the 2 Arduinos

arlimbad222, 2021-01-21:
Hey @SimonJohnCastle 

First sorry I can't understand that things because I am mechanical guy . Suppose I have one Uno with grbl 1.1 & second Uno blank which code I have to give second uno I don't have any experience on coding if it easy can you please make one for me .

SimonJohnCastle, 2021-01-21:
Get the Arduino IDE from [https://www.arduino.cc/en/software](https://www.arduino.cc/en/software)

The Arduino 'C' code is very simple, all you need is something like below, and upload it to your second Arduino, you can upload this and test it directly on your Uno from your PC.
If you open the Tools/Serial_Monitor you should see the output from the executing code, see image you need to select the speed 115200 which should match GRBL as they need to talk at the same bit speed (If all this is new to you)

`
void setup()
{
  Serial.begin(115200);
//  delay(10000); // Long 10 Second Delay to allow GRBL to BOOT and get ready
  Serial.println("G90");
  delay(100);
  Serial.println("G1 X10 Y10");
  delay(100);
  Serial.println("G1 X100");
  delay(1000); // Wait 1 Second for movment to complete
  Serial.println("G1 X-100");
  // ADD all your GCODE
}

void loop() { }
`

[image]

The above code will simply execute each time the Arduino resets, Press reset and see it run again in the open monitor.

arlimbad222, 2021-01-21:
Thankyou so much @SimonJohnCastle 

Now I understand is there any way to add push button to start , hold and stop cycle for machine safety.

SimonJohnCastle, 2021-01-21:
yes its very easy to add a button, once you see how easy it is you can learn on your own.
Look at the examples

arlimbad222, 2021-01-21:
Thanks again @SimonJohnCastle 

I'll buy electronics tomorrow then test it if I need any help I'll drop here 🙏

harryberlin, 2021-01-21:
check this 
https://github.com/mathertel/OneButton

arlimbad222, 2021-02-11:
hello @harryberlin @SimonJohnCastle 

I manage  all electronic today and test the code but when i give a code to mega with grbl it gives  lots of errors (25,24,1,2,) i think i want code that waits for ok response and then give second line of gcode to arduino mega with grbl & i am not good at coding side so please give me some advice i attach my ino file please see & give me advice 

have a good day thank you all for help
[M3M5.txt](https://github.com/gnea/grbl/files/5963185/M3M5.txt)

SimonJohnCastle, 2021-02-11:
My original idea for you, was to use a delay between each G-Code to allow the machine to move before the next command is sent.
This way do dont have to worry about the OK.
Start simple by just having one or two movements in your button press, to prove the concept.
i.e. does Grbl get out of lock when you reset and perform the first move when you press the button.
Then add in the extra G-Code with appropriate delays

arlimbad222, 2021-02-11:
HELLO @SimonJohnCastle 

oh i just forgot to use delay between command line i will check this as soon as possible & update here .

thanks for quick responce.

SimonJohnCastle, 2021-02-11:
you need something like an 11 second delay after the G01 Y22 F125
i.e. delay(11000);
would be interesting to know what the machine is doing ?

arlimbad222, 2021-02-11:
hi @SimonJohnCastle 

i am using one old horizontal milling machine to make 26 tooth on cast iron ring i attach some photos .

i am using  X as a rotary axis & Y as a linear axis . not using ball screw or any cnc stuff just put 2 stepper motor and boom its working .

[image]
[image]
[image]

SimonJohnCastle, 2021-02-11:
Nice, now I undestand the GCOED

arlimbad222, 2021-02-11:
Hi @SimonJohnCastle 

Sorry for same question but I have 3 products like this & I am to lazy to fine each code delay time I think there is some code that read ok message from grbl then put a second line of gcode if you have any idea how to do that please share I don't have any kind of coding experience.

Thankyou again and again .

SimonJohnCastle, 2021-02-11:
In principle its not too difficult, assuming you dont get into error conditions and want to make it fault tolerant.
The basic idea is to read from the serial port, either into an array or String and then look for the OK.
You keep doing this until the OK arrives and then move on to sending the next GCODE,
Look up
[Serial](https://www.arduino.cc/reference/en/language/functions/communication/serial/) see readString etc
[String](https://www.arduino.cc/reference/en/language/variables/data-types/stringobject/) see lastIndexOf
I cant git you a working example of the top of my head, but its not too difficult.

harryberlin, 2021-02-11:
G4 should be command for waiting

SimonJohnCastle, 2021-02-11:
This does what you need, Tested as shown, type ok and press send
Adjust it as needed for your button handling

`void setup()
{
  Serial.begin(115200);
  click1();
}

void sendWait(char * s)
{
  bool found = false;
  while (Serial.read() > 0); // read any previous junk
  
  Serial.println(s);

  while (!found)
  {
    while(Serial.available() < 2);
    if (Serial.read() == 'o' && Serial.peek() == 'k') found = true;
  }
}

void click1()
{
  sendWait("G91 G21");
  sendWait("G00 Y31");
  for (int i = 1; i <=26; i++)
  {
    sendWait("G01 Y22 F125");
    sendWait("G00 Y3");
    sendWait("G00 Y-25");
    sendWait("G01 X13.846 F5000");  
  }
  sendWait("G00 Y-31");
}

void loop()
{ }
`
[image]

arlimbad222, 2021-02-11:
Hi @SimonJohnCastle 

Thankyou for your efforts I will test code by tomorrow I am little busy today sorry for that I'll get back here when done something practical test .

Thankyou again and again brother .

arlimbad222, 2021-02-12:
hi @SimonJohnCastle 

I test the code but there is little problem when i want to send some real time command like (0x91 & 0x94) by button 2 . it finish that button 1 task then button 2 is working sorry for lots of question but seriously I don't have any coding skills i just copy and pest your code  .

have a nice day my friend .

SimonJohnCastle, 2021-02-12:
your string is wrong it should be "\x91" which means a single character of value hex 91.
Second I dont think GRBL responds with ok to real time commands so just use print or println like before, I dont know off my head if GRBLE needs the linefeed for relat time cmds