gnea/grbl #996: Skipping symbols sent by serial

Snow4DV, 2021-02-01:
Hello! By some reason i've got some letters missing in the serial monitor or ugs platform.
After compiling it for uno i also got the message that i'm low on memory (but that should be ok):
`Sketch uses 29766 byte (92%) memory of device. Mem available: 32256 byte.`
Here how it looks like sometimes when i send "?" command:
`<Idle|MPo:10.000,-1.500,0.000|FS:0,0>
ok
<Idle|MPs:10.000,-1.500,0.000|FS:0,0|WCO:0.000,0.000,0.000>
ok
Idle|MPos:10.000,-1.500,0.000|FS:0,0>
ok
<Idl|MPos:10.000,-1.500,0.000|FS:0,0>
ok
`

i tried 1.1h, 1.1f. 1.1h compiled with inverted switch - it remains the same.

PicEngraver, 2021-02-02:
https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl#status-reporting

**Status Reporting**

When a ? character is sent to Grbl (no additional line feed or carriage return character required), it will immediately respond with something like <Idle,MPos:0.000,0.000,0.000,WPos:0.000,0.000,0.000> to report its state and current position. The ? is always picked-off and removed from the serial receive buffer whenever Grbl detects one. So, these can be sent at any time. Also, to make it a little easier for GUIs to pick up on status reports, they are always encased by <> chevrons.

Developers can use this data to provide an on-screen position digital-read-out (DRO) for the user and/or to show the user a 3D position in a virtual workspace. We recommend querying Grbl for a ? real-time status report at no more than 5Hz. 10Hz may be possible, but at some point, there are diminishing returns and you are taxing Grbl's CPU more by asking it to generate and send a lot of position data.

Grbl's status report is fairly simply in organization. It always starts with a word describing the machine state like IDLE (descriptions of these are available elsewhere in the Wiki). The following data values are usually in the order listed below and separated by commas, but may not be in the exact order or printed at all. Report output depends on the user's $10 status report mask setting.

    MPos:0.000,0.000,0.000 : Machine position listed as X,Y,Z coordinates. Units (mm or inch) depends on $13 Grbl unit reporting setting.
    WPos:0.000,0.000,0.000 : Work position listed as X,Y,Z coordinates. Units (mm or inch) depends on $13 Grbl unit reporting setting.
    Buf:0 : Number of motions queued in Grbl's planner buffer.
    RX:0 : Number of characters queued in Grbl's serial RX receive buffer.

Snow4DV, 2021-02-03:
> https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl#status-reporting
>
> **Status Reporting**
>
> When a ? character is sent to Grbl (no additional line feed or carriage return character required), it [...]

I'm using universal gcode platform and status poll rate is 500 ms (on 250 it is the same). Here is the issue in their repo how it looks like: https://github.com/winder/Universal-G-Code-Sender/issues/1527#issuecomment-770774359
It doesn't look like it asks for position too often

SimonJohnCastle, 2021-02-03:
Can you say something about your HW, both PC/Grbl Is this new behavior did it work before.
What sort of USB1.1/2/3 is it an old PC, do you have anything else on the the same set of USB ports.
If you are using one of those woodpecker boards, dont have the offline controller connected at the same time.
Is it always the same missing characters etc

Snow4DV, 2021-02-12:
> Can you say something about your HW, both PC/Grbl Is this new behavior did it work before.
> What sort of USB1.1/2/3 is it an old PC, do you have anything else on the the same set of USB ports.
> If you are using one of those woodpecker boards, dont have the offline controller connected at the same time.
> Is it always the same missing characters etc
> Which Arduino IDE version
> Do you have possible ground loop issues

Sry for late answer. Just a usual not very old pc  - i5 8400/8gb ram. Tried both usb 3.1 and 2.0 - same results. Yeah - pheripherials (keyboard n mouse l, stm32 (hid). 

I am using a plain uno + v3 cnc shield + drv8825x2 + servo connected to z endstop

Missing chars are always different 
Using arduino ide 1.8.9 

Ground loops are possible because i have psu 5v/30v and pc 5v at the same time. Can this be an issue? 

Thx! :)

SimonJohnCastle, 2021-02-12:
Normally you need all the GNDs connected together, but dont try and drive the digital side via more than one 5V source.
Given the fact you managed to flash it suggests the arduino is fine, try and run it without the shield and mechanical HW.
You can run a full milling sequence on a bare arduino so long as its not expecting to do homing etc.
what is the stm32, some sort of home brew joystick or something ? could this be flooding the USB ? try removing it.
Depending on what you are doing with the 30V if its separate (spindle motor) etc you may need an opto isolator and 
completely separate it.

Snow4DV, 2021-02-15:
> Normally you need all the GNDs connected together, but dont try and drive the digital side via more than one 5V source.
> Given the fact you managed to flash it suggests the arduino is fine, try and run it without the shield and mechanical HW.
> You can run a full milling sequence on a bare arduino so long as its not expecting to do homing etc.
> what is the stm32, some sort of home brew joystick or something ? could this be flooding the USB ? try removing it.
> Depending on what you are doing with the 30V if its separate (spindle motor) etc you may need an opto isolator and
> completely separate it. Im running a woodpeker board that takes its 5v from the PC so runs fine when the CNC3018 is powered of, so you should not need extra 5V unless you are worried about the servo, but you can give it its own 5V and just take common GND and signal from the arduino

Ok! Will give it a try soon. Stm32 - is force feedback steering wheel and i know that it's sending some axis states not through the hid but by its own protocol - didn't really think about it, got 2 possible causes of this behaviour,