gnea/grbl #997: Y and Z axis issues

Beewap, 2021-02-01:
Hi,

I'm new into cnc GRBL world and I have a 3018 pro cnc (from Mostics) which looks really cool but I have two issues with Y and Z axis.

With the Z axis, when I push the "Home" button, the head goes up in the Z axis, hit the limit and still try to go upper. So the motor does some noises and after few second of trying it stop and candle lockup everything.
I try to set up the $132 with lower value like 15. And even if I put the bit at the lowest position when I hit the home button it still try to go upper. But with this low value it stop just before the hit the stop but it doesn't go to 0 on the x and y axis.
Also, when I turn on the router it's already locked.. 

Second issue  on the Y axis now.
I wanted to engrave a picture, I set up the file with easel and the preview was good. Nothing special.
Into candle, the file looks good in the preview window, I setup the workspace area and I start the engraving. It was fine but at one moment the Y axis has an issue. It's look like the router can't or wont go where it should (but it doesn't stop) and continue the engraving with is path modified.

Here is a picture of the issue. The orange dot is the true limit of the shape it should do but the router goes, at every turn, lower on the Y axis. (Don't pay attention to top line it was a previous test).

[image]

Here is my config :
$1=25
$2=0
$3=3
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=1
$23=0
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=255
$31=0
$32=1
$100=800.000
$101=800.000
$102=800.000
$110=800.000
$111=800.000
$112=600.000
$120=300.000
$121=300.000
$122=30.000
$130=260.000
$131=150.000
$132=15.000

Any idea ?

PicEngraver, 2021-02-02:
If you are using a Woodpecker ver. 3.4 board, the screen print markings for the X and Z axis are reversed, which will make your homing not work.

Make sure your carving is centered and not larger than your machine's travel limits ($130/$131/$132)

Beewap, 2021-02-02:
Yes it is a Woodpecker 3.4 board but the carving move correctly. When I go left on candle the carving goes left, I go forward it goes forward and if I go up on Z axis, the carving goes up too. So it's looks like it move as it should.

I don't understand how/why the carvin, in the middle of the carving, there is this issue of moving on Y axis way too much. :/

There is something on my $$ command that look wierd ?

PicEngraver, 2021-02-02:
Your report of your homing failure indicates that your switches are not working correctly.  See my first reply.  The switches have nothing to do with axis movements.   You have homing enabled, so your startup will always be "locked" until you complete the homing cycle, or unlock grbl.  Unlocking is not recommended unless necessary because it can cause other issues.  Fix the homing issue before you try to proceed, or set homing off in the grbl settings ($22=0).   I am assuming that you do have a set of homing switches installed.  This is not clear from your first message.  If you do not have switches, then for sure set $22=0.

You are obviously losing Y axis position.  This is likely caused by the Y hitting the hard stop of your machine.  Either reposition your carving up to avoid hitting a hard stop (max. physical position of the table), or assure that the image is small enough to carve within the physical movement limits of your machine.

Beewap, 2021-02-02:
If switch limit is a physical sensor that stop the head when it touch the sensor, so I don't have switchs. I'll change $22=0.

As you say, I have an issue with losing Y axis position but  I'm still far away the physical limit of the table, few centimetres left.
Maybe I wasn't clear on the picture, the carving under the orange dot are not welcome. :p So the engraving was free to go forward but still going backward on the wood. (and I had still few centimetres backward before hit the physical limits.

About the $100/$101/$102, it's possible that these values are incorrect. It was stock value and I haven't calibrated my machine. I will do that later today.

PicEngraver, 2021-02-02:
You may have the feed rate too high.  Try slowing down.  Also settings $110/$111 (max feed rate) and $120/$121 (axis acceleration) may be a little too high.  Try with them reduced, too.

Beewap, 2021-02-02:
Come back with few quick questions.

I have calibrated the $100/$101/$102 (it was close to be good on stock value). Also $130/$131/$132 for the limit.

Also I wanna know if there is some logical way for the head to move in the good direction.
With the head in the left top corner of the table, :
- if I wanna go to the right it will be a positive coordinate (x axis)
- if I wanna go to the bottom it will be a negative coordinate (y axis)
- if I wanna go lower (so the head come closer to the table) it will be a negative coordinate (z axis)

Is the X axis inverted ? That's what your are talking about earlier ? It had to be set up in $2=1 (x axis inverted).

So now my cnc is calibrated and, if needed, my X axis can be inverted. What's the good way now to start an engraving ?
Puti the head on 0,0,0 on the top left corner  and Z axis in his higher position ?
Put it in the center of the table ?

Also, if I try to move larger than is physical limit candle will still thinks it moving, so if I was already in 0 on X axis for exemple, and it try to move to far on left (negative value) in candle the coordinate will be negative instead of 0 even if it dind't and can't move. And if I send after G0 X0 command the head will not go in 0 but 0+ the distance it think it was before. 
Don't know if what I'm saying is understandable. Let me now. :)

PicEngraver, 2021-02-03:
It is obvious that you have many, many questions about cnc operations in general, and also questions about basic Candle and grbl control.  This, IMO, is not the place to get all your questions answered.  I do not use Candle, so can't help with that anyway.

Better, I think, if you look into these links for specific questions about your machine and its setup.  I'm sure a lot of your questions will be answered far better than I can.
https://docs.sainsmart.com/
https://www.facebook.com/groups/SainSmart.GenmitsuCNC/
https://docs.sainsmart.com/article/3jgtgas880-genmitsu-3018-pro-resources

Best to you,
John

Beewap, 2021-02-04:
Hi, 

Yes, I understand, thank you for your help anyway and these links. Already read it yesterday s I have a better understanding now.
Thanks !